zachjs/sv2v #123: generate if/else colliding with always if/else

Mop-u, 2021-01-25:
build ver: `v0.0.6-9-gb28a3ca`
It seems that there may be some edge cases where begin/end are being stripped away too aggressively. The issue can be worked around for now by preventing begin/end from being synth'd away using `begin : block_tag`
without that guarding, this code gets a syntax error after conversion

input hdl:
```verilog
module Lfsr #(
    parameter depth = 16,
    parameter [depth-1:0] xor_map  = 16'h8016,
    parameter [depth-1:0] init_map = 16'hBEEF
)(
    input clk,
    input rst,
    input Step,
    output LfsrOut
);
localparam zeroed_map = !init_map;
genvar i;
generate
    reg [depth-1:0] ShiftRegs;
    wire Feedback = ^(xor_map & ShiftRegs);
    for(i=0;i<depth;i=i+1) begin : gen_lfsr_loop
        if(i==0) begin
            always_ff @(posedge clk, posedge rst) begin
                if(rst)       ShiftRegs[0] <= zeroed_map ? 1 : init_map[0];
                else if(Step) ShiftRegs[0] <= Feedback;
            end
        end
        else begin
            always_ff @(posedge clk, posedge rst) begin
                if(rst)       ShiftRegs[i] <= init_map[i];
                else if(Step) ShiftRegs[i] <= ShiftRegs[i-1];
            end
        end
    end
    assign LfsrOut = ShiftRegs[depth-1];
endgenerate
endmodule
```
broken output, resulting in a syntax error on the second always block:
```verilog
module Lfsr (
	clk,
	rst,
	Step,
	LfsrOut
);
	parameter depth = 16;
	parameter [depth - 1:0] xor_map = 16'h8016;
	parameter [depth - 1:0] init_map = 16'hbeef;
	input clk;
	input rst;
	input Step;
	output LfsrOut;
	localparam zeroed_map = !init_map;
	genvar i;
	generate
		reg [depth - 1:0] ShiftRegs;
		wire Feedback = ^(xor_map & ShiftRegs);
		for (i = 0; i < depth; i = i + 1) begin : gen_lfsr_loop
			if (i == 0) always @(posedge clk or posedge rst)
				if (rst)
					ShiftRegs[0] <= (zeroed_map ? 1 : init_map[0]);
				else if (Step)
					ShiftRegs[0] <= Feedback;
			else always @(posedge clk or posedge rst)
				if (rst)
					ShiftRegs[i] <= init_map[i];
				else if (Step)
					ShiftRegs[i] <= ShiftRegs[i - 1];
		end
		assign LfsrOut = ShiftRegs[depth - 1];
	endgenerate
endmodule
```

patched input with begin/end guarded with tags:
```verilog
module Lfsr #(
    parameter depth = 16,
    parameter [depth-1:0] xor_map  = 16'h8016,
    parameter [depth-1:0] init_map = 16'hBEEF
)(
    input clk,
    input rst,
    input Step,
    output LfsrOut
);
localparam zeroed_map = !init_map;
genvar i;
generate
    reg [depth-1:0] ShiftRegs;
    wire Feedback = ^(xor_map & ShiftRegs);
    for(i=0;i<depth;i=i+1) begin : gen_lfsr_loop
        if(i==0) begin
            always_ff @(posedge clk, posedge rst) begin : gen_lfsr_feedback
                if(rst)       ShiftRegs[0] <= zeroed_map ? 1 : init_map[0];
                else if(Step) ShiftRegs[0] <= Feedback;
            end
        end
        else begin
            always_ff @(posedge clk, posedge rst) begin : gen_rest_of_lfsr
                if(rst)       ShiftRegs[i] <= init_map[i];
                else if(Step) ShiftRegs[i] <= ShiftRegs[i-1];
            end
        end
    end
    assign LfsrOut = ShiftRegs[depth-1];
endgenerate
endmodule
```
correct output:
```verilog
module Lfsr (
	clk,
	rst,
	Step,
	LfsrOut
);
	parameter depth = 16;
	parameter [depth - 1:0] xor_map = 16'h8016;
	parameter [depth - 1:0] init_map = 16'hbeef;
	input clk;
	input rst;
	input Step;
	output LfsrOut;
	localparam zeroed_map = !init_map;
	genvar i;
	generate
		reg [depth - 1:0] ShiftRegs;
		wire Feedback = ^(xor_map & ShiftRegs);
		for (i = 0; i < depth; i = i + 1) begin : gen_lfsr_loop
			if (i == 0) always @(posedge clk or posedge rst) begin : gen_lfsr_feedback
				if (rst)
					ShiftRegs[0] <= (zeroed_map ? 1 : init_map[0]);
				else if (Step)
					ShiftRegs[0] <= Feedback;
			end
			else always @(posedge clk or posedge rst) begin : gen_rest_of_lfsr
				if (rst)
					ShiftRegs[i] <= init_map[i];
				else if (Step)
					ShiftRegs[i] <= ShiftRegs[i - 1];
			end
		end
		assign LfsrOut = ShiftRegs[depth - 1];
	endgenerate
endmodule
```

zachjs, 2021-01-25:
Thank you for catching this issue and reporting it! It should now be fixed. Please let me know if the issue persists.

Mop-u, 2021-01-25:
just checked & this fixes both files I was having issues with, thanks!

closed 2021-01-25 by Mop-u