zachjs/sv2v #124: duplicate wire declaration

Mop-u, 2021-01-25:
Build ver: `v0.0.6-9-gb28a3ca`
The following code ends up generating a duplicate wire declaration for `BusOutOdd` that's only 1 bit wide.
If I put the declaration before the module it's feeding, the issue disappears.

input file:
```verilog
module BusCombiner #(
    parameter bwidth = 1600,
    parameter bcount = 4,
    parameter speed_opt = 1
)(
    input [bcount-1:0][bwidth-1:0] BusesIn,
    output [bwidth-1:0] BusOut
);
parameter odd = bcount&1;
generate
    if(bcount==1) begin
        assign BusOut = BusesIn;
    end
    else if(|speed_opt) begin : sv2v_keep_this_begin
        if(odd) begin
            BusCombiner #(.bwidth(bwidth),.bcount(bcount-1),.speed_opt(1))
            bus_comb_odd (
                .BusesIn(BusesIn[bcount-2:0]),
                .BusOut (BusOutOdd)
            );
            assign BusOut = BusOutOdd|BusesIn[bcount-1];
            wire [bwidth-1:0] BusOutOdd;
        end
        else if(bcount>2) begin
            wire [bwidth-1:0] BusOutA;
            wire [bwidth-1:0] BusOutB;
            BusCombiner #(.bwidth(bwidth),.bcount(bcount/2),.speed_opt(1))
            bus_comb_A (
                .BusesIn(BusesIn[bcount-1:(bcount/2)]),
                .BusOut (BusOutA)
            );
            BusCombiner #(.bwidth(bwidth),.bcount(bcount/2),.speed_opt(1))
            bus_comb_B (
                .BusesIn(BusesIn[(bcount/2)-1:0]),
                .BusOut (BusOutB)
            );
            assign BusOut = BusOutA|BusOutB;
        end
        else begin
            assign BusOut = BusesIn[1]|BusesIn[0];
        end
    end
    else begin
        wire [bcount-1:0][bwidth-1:0] BusComb;
        genvar i;
        assign BusComb[bcount-1] = BusesIn[bcount-1];
        for(i=0;i<bcount-1;i=i+1) begin : area_opt_combine
            assign BusComb[i] = BusesIn[i]|BusComb[i+1];
        end
        assign BusOut = BusComb[0];
    end
endgenerate
endmodule
```

generated output with duplicate `BusOutOdd` declaration:
```verilog
module BusCombiner (
	BusesIn,
	BusOut
);
	parameter bwidth = 1600;
	parameter bcount = 4;
	parameter speed_opt = 1;
	input [(bcount * bwidth) - 1:0] BusesIn;
	output [bwidth - 1:0] BusOut;
	parameter odd = bcount & 1;
	generate
		if (bcount == 1) assign BusOut = BusesIn;
		else if (|speed_opt) begin : sv2v_keep_this_begin
			if (odd) begin
				wire BusOutOdd;
				BusCombiner #(
					.bwidth(bwidth),
					.bcount(bcount - 1),
					.speed_opt(1)
				) bus_comb_odd(
					.BusesIn(BusesIn[bwidth * (((bcount - 2) >= 0 ? bcount - 2 : ((bcount - 2) + ((bcount - 2) >= 0 ? bcount - 1 : 3 - bcount)) - 1) - (((bcount - 2) >= 0 ? bcount - 1 : 3 - bcount) - 1))+:bwidth * ((bcount - 2) >= 0 ? bcount - 1 : 3 - bcount)]),
					.BusOut(BusOutOdd)
				);
				assign BusOut = BusOutOdd | BusesIn[(bcount - 1) * bwidth+:bwidth];
				wire [bwidth - 1:0] BusOutOdd;
			end
			else if (bcount > 2) begin
				wire [bwidth - 1:0] BusOutA;
				wire [bwidth - 1:0] BusOutB;
				BusCombiner #(
					.bwidth(bwidth),
					.bcount(bcount / 2),
					.speed_opt(1)
				) bus_comb_A(
					.BusesIn(BusesIn[bwidth * (((bcount - 1) >= (bcount / 2) ? bcount - 1 : ((bcount - 1) + ((bcount - 1) >= (bcount / 2) ? ((bcount - 1) - (bcount / 2)) + 1 : ((bcount / 2) - (bcount - 1)) + 1)) - 1) - (((bcount - 1) >= (bcount / 2) ? ((bcount - 1) - (bcount / 2)) + 1 : ((bcount / 2) - (bcount - 1)) + 1) - 1))+:bwidth * ((bcount - 1) >= (bcount / 2) ? ((bcount - 1) - (bcount / 2)) + 1 : ((bcount / 2) - (bcount - 1)) + 1)]),
					.BusOut(BusOutA)
				);
				BusCombiner #(
					.bwidth(bwidth),
					.bcount(bcount / 2),
					.speed_opt(1)
				) bus_comb_B(
					.BusesIn(BusesIn[bwidth * (((bcount / 2) - 1) - ((bcount / 2) - 1))+:bwidth * (bcount / 2)]),
					.BusOut(BusOutB)
				);
				assign BusOut = BusOutA | BusOutB;
			end
			else assign BusOut = BusesIn[bwidth+:bwidth] | BusesIn[0+:bwidth];
		end
		else begin
			wire [(bcount * bwidth) - 1:0] BusComb;
			genvar i;
			assign BusComb[(bcount - 1) * bwidth+:bwidth] = BusesIn[(bcount - 1) * bwidth+:bwidth];
			for (i = 0; i < (bcount - 1); i = i + 1) begin : area_opt_combine
				assign BusComb[i * bwidth+:bwidth] = BusesIn[i * bwidth+:bwidth] | BusComb[(i + 1) * bwidth+:bwidth];
			end
			assign BusOut = BusComb[0+:bwidth];
		end
	endgenerate
endmodule

```

fixed input snippet:
```verilog
        if(odd) begin
            wire [bwidth-1:0] BusOutOdd;
            BusCombiner #(.bwidth(bwidth),.bcount(bcount-1),.speed_opt(1))
            bus_comb_odd (
                .BusesIn(BusesIn[bcount-2:0]),
                .BusOut (BusOutOdd)
            );
            assign BusOut = BusOutOdd|BusesIn[bcount-1];
        end
```
fixed generated output snippet:
```verilog
			if (odd) begin
				wire [bwidth - 1:0] BusOutOdd;
				BusCombiner #(
					.bwidth(bwidth),
					.bcount(bcount - 1),
					.speed_opt(1)
				) bus_comb_odd(
					.BusesIn(BusesIn[bwidth * (((bcount - 2) >= 0 ? bcount - 2 : ((bcount - 2) + ((bcount - 2) >= 0 ? bcount - 1 : 3 - bcount)) - 1) - (((bcount - 2) >= 0 ? bcount - 1 : 3 - bcount) - 1))+:bwidth * ((bcount - 2) >= 0 ? bcount - 1 : 3 - bcount)]),
					.BusOut(BusOutOdd)
				);
				assign BusOut = BusOutOdd | BusesIn[(bcount - 1) * bwidth+:bwidth];
			end
```

zachjs, 2021-01-25:
This rather odd output results from sv2v seeing the use of `BusOutOdd` before it has been declared, and because the `default_nettype` is `wire`, it creates the implicit net ahead of that first use.

Section 6.5 of IEEE 1800-2017 states

> Data shall be declared before they are used, apart from implicit nets (see 6.10).

Some tools (commercial or otherwise) enforce this rule. sv2v includes an initial phase for reordering declarations to fix sources violating this rule, but it currently only performs this transformation for declarations in packages and those at the top level of modules. I would like to extend this support to include declarations in generate blocks. You can expect an update on this soon.

Mop-u, 2021-01-25:
yep, issue resolved, thanks!

closed 2021-01-25 by Mop-u