zachjs/sv2v #125: Yosys error with multiple wrapped casts

nmoroze, 2021-01-30:
Hi! If I run sv2v on the following SystemVerilog example:

```verilog
module example  (
  input [1:0] in,
);

  typedef struct packed {
    logic [1:0]  a;
  } mystruct;

  mystruct out;

  assign out = '{
    a:  2'(in[1:0])
  };

endmodule
```

with`sv2v example.sv > example.v` followed by `yosys -p 'read_verilog example.v'` I get the following error: "ERROR: Non-constant function call in constant expression." However, I'd expect the Yosys front-end to parse the Verilog without any issues.

I noticed that the issue itself seems to be due to having multiple wrapped sv2v cast functions, in particular the following line in the output: `assign out = sv2v_cast_2(sv2v_cast_2(in[1:0]));`. Removing the second `sv2v_cast_2` by hand makes Yosys parse the output fine. 

A few additional observations I made while debugging:
- This problem only happens with a struct assignment
- This problem only happens with an explicit width on the RHS. It seems a bit redundant in this example, but this is reduced from some code I found out in the wild (https://github.com/lowRISC/opentitan/blob/master/hw/vendor/lowrisc_ibex/rtl/ibex_cs_registers.sv#L510) 
- This problem does not occur in versions of Yosys prior to commit yosyshq/yosys@2085d9a55d38f28551739d63bede29d93edaf578

I'm honestly not sure if this should be considered an issue in sv2v or Yosys (the extra function wrapping seems unnecessary, but not incorrect to me). However, since you wrote the commit that seems to introduce this problem into Yosys, I figured you might have some good insight either way!

Thanks in advance for taking a look!

zachjs, 2021-01-30:
Thanks for the detailed report! This is indeed a regression I caused in Yosys (sorry!), but the appropriate fixes are already pending. See YosysHQ/Yosys#2562 which contains the larger discussion so far.

I am interested in taking a pass at the cast conversion to try to remove some of the unnecessary cast function calls. I believe there is at least one instance of a double function call of this style _not_ due to the conversion in OpenTitan (not sure if it's in the Ibex core), so I suggest using my Yosys `repeat-call` branch for the time being.

I'd like to leave this issue open to track progress on the cast conversion.

nmoroze, 2021-02-01:
Ah I see, I guess I should have checked the Yosys issues more closely first! Thanks as always for the prompt reply and for pointing me towards the fixes.

zachjs, 2021-02-11:
The Yosys constant function evaluation issue should now be fixed on master. sv2v should also be producing substantially fewer unnecessary cast functions, though I believe there is still room for improvement. Please let me know what you find!

nmoroze, 2021-02-22:
An unrelated Yosys issue (exact problem TBD) has been keeping me from using the latest Yosys with your patch, but it does look like the extra cast mentioned in this issue no longer appears. Everything works for me now even on the older version of Yosys!

zachjs, 2021-03-21:
It sounds like this issue has been resolved. Thanks again!

closed 2021-03-21 by zachjs