zachjs/sv2v #126: Interfaces and empty output

ben-marshall, 2021-01-31:
Hi there
I'm having some trouble with translating (what I thought was) a basic interface and top level module. I've tried the example code below on both the pre-compiled `v0.0.6` Linux release, and a freshly compiled version of HEAD at 275130e0.

I was expecting the interface to be expanded, but instead I get empty output. Adding the `-v` flag tells me all of the constructs are simply removed.

Input file:
```systemverilog
interface ex_if_csrs (
    input   g_clk   ,
    input   g_resetn
);

parameter   XLEN = 32       ;
localparam  XL   = XLEN-1   ;

logic        hvalid  ;
logic [ 5:0] haddr   ;
logic        hwen    ;
logic        hset    ;
logic        hclr    ;
logic [XL:0] hwdata  ;
logic [XL:0] erdata  ;
logic        eready  ;

modport HOST (
output hvalid  ,
output haddr   ,
output hwen    ,
output hset    ,
output hclr    ,
output hwdata  ,
input  erdata  ,
input  eready
);

modport EX (
input  hvalid  ,
input  haddr   ,
input  hwen    ,
input  hset    ,
input  hclr    ,
input  hwdata  ,
output erdata  ,
output eready
);

endinterface

module ex_top #(
parameter XLEN  =   32  ,
parameter VLEN  =   128
)(
input   wire    g_clk       ,
input   wire    g_clk_req   ,
input   wire    g_resetn    ,

ex_if_csrs.EX   csrif

);

localparam  XL   = XLEN-1   ;

assign csrif.eready = 1'b1;

endmodule

```
Command line + output:
```
ben@ben-TF:~/projects/riscv-exmoor$ cat /tmp/test.sv | /opt/eda/sv2v/v0.0.6/sv2v -v -
// removed interface: ex_if_csrs
// removed module with interface ports: ex_top
```

I'm really sorry if it turns out I'm missing something obvious, I've been scratching my head for an hour now!

Many thanks,
Ben

zachjs, 2021-01-31:
Without verbose mode, sv2v produces the following warning to stderr with the given example:
> Warning: Source includes an interface but output is empty because there is no top-level module which has no ports which are interfaces.

I will make these warnings appear in verbose mode too.

sv2v converts interfaces and modules with interface ports by inlining them. While some simple interfaces can be elaborated into bundles of ports, this is not always the case. In general, the top level module cannot itself have interface ports. Many tools produce errors or warnings for the given example for this reason.

zachjs, 2021-02-03:
> Without verbose mode, sv2v produces the following warning to stderr with the given example:
> 
> > Warning: Source includes an interface but output is empty because there is no top-level module which has no ports which are interfaces.

I have fixed sv2v to correctly produce the above warning in verbose mode when applicable.

ben-marshall, 2021-02-03:
> I have fixed sv2v to correctly produce the above warning in verbose mode when applicable.

Ah, that's great thank you.

So, the bottom line is that, like other tools, sv2v doesn't do conversion of interfaces for the top-level module in an design? If I want to use sv2v for a design which _does_ have interfaces at the top-level, then creating a dummy wrapper which manually expands everything would be a way around it?

zachjs, 2021-02-03:
> So, the bottom line is that, like other tools, sv2v doesn't do conversion of interfaces for the top-level module in an design?

sv2v doesn't have any special logic for detecting or dealing with top-level modules. The interface conversion works by inlining both interfaces and modules with interface ports. Because they are expected to be inlined, those original elements are dropped.

> If I want to use sv2v for a design which _does_ have interfaces at the top-level, then creating a dummy wrapper which manually expands everything would be a way around it?

If there was a wrapper around `ex_top`, the interface would indeed be converted, but sv2v would do so by inlining `ex_top` into the wrapper. Unfortunately, I'm not aware of any standard "ABI" for how top-level modports should be exposed.

ben-marshall, 2021-02-10:
Thanks for your explanations, they're really helpful. I'll close this issue since there's no "fix" to be done here. It's a great tool, I'm really glad it exists!

closed 2021-02-10 by ben-marshall

zachjs, 2021-02-10:
Thanks for the feedback! If you have suggestions for how sv2v can reduce friction in common use cases, please don't hesitate to file new issues.