zachjs/sv2v #127: '0 / '1 replication issue with nested interface

Mop-u, 2021-02-05:
I think there may be some scoping issue when converting the sv style bit fill to a constant bit replication when using interfaces.

The issue shows up in the last 2 assignments of the Regfile module - if '0 is changed to 0, the testbench compiles and runs correctly. Otherwise the following errors are reported by iverilog:
```
.//../tmp/testbench.sv:74: error: Unable to bind parameter `_tmp_4AFE0_data_w' in `testbench.regfile'
.//../tmp/testbench.sv:75: error: Unable to bind parameter `_tmp_4AFE0_data_w' in `testbench.regfile'
```

I've stripped away as much as I can of my code to just the involved modules / interfaces:

```verilog
module testbench ();

reg rst=0, clk=0;
initial begin
    #10 rst = 1;
    #10 rst = 0;
end
always #100 clk = ~clk;

mem_intf #(
    .read_ports (2),
    .write_ports(1),
    .addr_w     (5),
    .data_w     (32)
) RfIntf();

integer Cycle = 0;
integer Stop = 4;//2048;
always_ff @(posedge clk, posedge rst) begin
    if     (rst)         Cycle <= 0;
    else if(Cycle>=Stop) $finish();
    else                 Cycle <= Cycle + 1;
end
always_ff @(posedge clk) begin
    $display("\nCycle %0d:",Cycle);
    $display("x%d => %h",RfIntf.Read[0].Addr, RfIntf.Read[0].Data);
    $display("x%d => %h",RfIntf.Read[1].Addr, RfIntf.Read[1].Data);
    $display("x%d <= %h",RfIntf.Write[0].Addr,RfIntf.Write[0].Data);
end

assign RfIntf.Read[0].Enable = 1;
assign RfIntf.Read[1].Enable = 1;
assign RfIntf.Write[0].Enable = 1;
assign RfIntf.Read[0].Addr = 'ha;
assign RfIntf.Read[1].Addr = 'ha;
assign RfIntf.Write[0].Addr = 'ha;
assign RfIntf.Write[0].Data = 'hb;
Regfile #(.addr_w(5),.data_w(32))
regfile (
    .clk   (clk),
    .RfIntf(RfIntf)
);

endmodule
```

```verilog
module Regfile #(
    parameter addr_w = 5,
    parameter data_w = 32
)(
    input clk,
    mem_intf RfIntf
);
reg [data_w-1:0] Registers [1:(2**addr_w)-1];
always_ff @(posedge clk) begin
    if(RfIntf.Write[0].Enable & |RfIntf.Write[0].Addr) begin
        Registers[RfIntf.Write[0].Addr] <= RfIntf.Write[0].Data;
    end
end
assign RfIntf.Read[0].Data = |RfIntf.Read[0].Addr ? Registers[RfIntf.Read[0].Addr] : '0;
assign RfIntf.Read[1].Data = |RfIntf.Read[1].Addr ? Registers[RfIntf.Read[1].Addr] : '0;
endmodule
```

```verilog
interface mem_intf #(
    parameter read_ports = 2,
    parameter write_ports = 1,
    parameter addr_w = 5,
    parameter data_w = 32
)();

wr_intf #(.addr_w(addr_w),.data_w(data_w)) Write [write_ports] ();
rd_intf #(.addr_w(addr_w),.data_w(data_w)) Read  [read_ports]  ();

endinterface
```
```verilog
// simple addr/data read bundle
interface rd_intf #(
    parameter addr_w = 5,
    parameter data_w = 32
)();
wire [addr_w-1:0] Addr;
wire [data_w-1:0] Data;
wire              Enable;
modport to_mem (
    output Addr,
    input  Data,
    output Enable
);
modport from_mem (
    input  Addr,
    output Data,
    output Enable
);
endinterface
```
```verilog
// simple addr/data write bundle
interface wr_intf #(
    parameter addr_w = 5,
    parameter data_w = 32
)();
wire [addr_w-1:0] Addr;
wire [data_w-1:0] Data;
wire              Enable;
modport to_mem (
    output Addr,
    output Data,
    output Enable
);
modport from_mem (
    input Addr,
    input Data,
    input Enable
);
endinterface
```

zachjs, 2021-02-07:
Thank you for providing a detailed bug report! The conversion of "unbased unsized literals" often depends on the type (size) of adjacent expressions. For this example, the resolution of that type information results in the size referring to an identifier which is out of scope. This "blind substitution" can come up in a number of different scenarios. Unfortunately, the LRM does not allow for constant expressions to contain hierarchical references. However, I have a decent idea of I can work around this limitation. You can expect a progress update in a few days.

zachjs, 2021-02-10:
Resolution of type information across scopes now correctly resolves, disambiguates, and substitutes constants. This specific example should now be converted correctly. While issues with "blind substitution" should be relatively rare, it is likely still possible to encounter them when dealing with structs/unions and arrays which are flattened. I will look at addressing those in the future, though concrete examples (if you encounter issues) are always appreciated.

zachjs, 2021-02-22:
This "blind substitution" should now also be resolved for all remaining conversions where it could have arisen: struct, typedef, and array flattening.

Mop-u, 2021-02-24:
cool! I'll close the issue, things are working smoothly on my end, thanks!

closed 2021-02-24 by Mop-u