zachjs/sv2v #128: sv2v: invalid based literal digits

nagendragd, 2021-02-14:
When I run the attached file through sv2v, I get the error:

 sv2v: invalid based literal digits: (2,25,268435455,268435455,28,25)
CallStack (from HasCallStack):
  error, called at src/Language/SystemVerilog/AST/Number.hs:239:9 in main:Language.SystemVerilog.AST.Number

Invocation of sv2v is as:
 sv2v \
    --define=SYNTHESIS \
    ../rtl/*_pkg.sv \
    -I../vendor/lowrisc_ip/ip/prim/rtl \
    -I../dv/fcov \
    $file \
    > $LR_SYNTH_OUT_DIR/generated/${module}.v

Any clues on what this is due to?

I pulled the latest sv2v code off github and installed it:
sv2v --version
sv2v v0.0.6-31-g8e1f2bb

Thanks
Nagendra
[ibex_tracer.sv.txt](https://github.com/zachjs/sv2v/files/5978696/ibex_tracer.sv.txt)

zachjs, 2021-02-15:
Thanks for reporting this issue! My recent addition of constant folding for concatenations of number literals did not correctly handle literals where the leading X/Z digit is implicitly truncated. It should now be fixed. Please let me know if it's working for you.

nagendragd, 2021-02-15:
Thanks Zach, issue fixed now! Much appreciated!

closed 2021-02-15 by nagendragd