zachjs/sv2v #130: Parse error: no tokens left to parse.

iamrishabh23, 2021-03-06:
Hi , 
I am trying to convert two files from System Verilog to Verilog and I face this error.
**Parse error: no tokens left to parse.**
I looked up and found no info regarding this error . 

The command I am using is as follows
sv2v --write=adjacent file1.sv file2.sv
 I have many System Verilog related Tokens in the code as in structs , logic etc.

Please let me know how to get past this issue.
Thanks.

zachjs, 2021-03-06:
Can you share the files you are attempting to convert? I can only effectively debug an error like this if I am able to reproduce it.

iamrishabh23, 2021-03-06:
There are 2 files , BasicTypes.sv containing typedefs inside a package and the second one Basic_module.sv ,has a module which uses the typedefs.
[Basic.zip](https://github.com/zachjs/sv2v/files/6095058/Basic.zip)

zachjs, 2021-03-06:
The second file is missing an `endmodule`. I am looking into ways of improving sv2v's error messaging here.

zachjs, 2021-03-06:
I've pushed improved error messaging for this and closely related syntax errors. The given inputs now produce the following error:
```
Basic/Basic_module.sv:33:1: Parse error: missing expected `endmodule`
```

iamrishabh23, 2021-03-08:
Oh that was so silly of me !

closed 2021-03-08 by iamrishabh23

zachjs, 2021-03-08:
No worries! I'm glad I could help. This was a good motivator to improve some of sv2v's error messaging. I think many other users stand to benefit. Thank you!