zachjs/sv2v #131: Upload package to Hackage

QuantamHD, 2021-03-09:
Hi @zachjs,

Would it be possible to upload this package hackage? It would make importing this code into bazel which is an external build system for haskell.

Thanks

zachjs, 2021-03-10:
Thank you for your interest in the project! I've begun this process, but expect it may take at least a few days. I'll give an update soon.

QuantamHD, 2021-03-12:
Thanks for providing such great software! I really appreciate the help

zachjs, 2021-03-19:
I have created a new release, v0.0.7, and a corresponding _candidate_ release on Hackage: https://hackage.haskell.org/package/sv2v-0.0.7/candidate.

zachjs, 2021-04-09:
Have you had a chance to try out the uploaded candidate release?

QuantamHD, 2021-04-15:
It's working great! Thank you Zach

zachjs, 2021-04-15:
I noticed an issue with that candidate. The tarball naturally doesn't contain the git repo, which breaks a compile-time dependency on the output of `git describe`. I'm curious what you get when you run `sv2v --version` on the binary you've installed this way. Is it the `git describe` output of some outer repo?

I fixed this issue in dc19b5f9442931ea9e6df4d4d5f25b7b387e2cef, so it should work correctly in the next release.

Because of the issue, I will be deleting that candidate in the long run, but not before I create and release a new version.

zachjs, 2021-08-01:
The first non-candidate release of sv2v on Hackage is now available: https://hackage.haskell.org/package/sv2v. I have confirmed that `cabal install sv2v` now works as expected. I'm marking this as done for now, but please feel free to follow up if necessary!

closed 2021-08-01 by zachjs