zachjs/sv2v #134: Enums of custom types in package

vogelpi, 2021-04-09:
Hi,

while playing around with the latest version of sv2v (v0.0.7-3-g2a4d1cc) I noticed that the tool seems to have an issue with enum of custom types defined in packages. For example, I have a package `lc_ctrl_pgk.sv` which contains:
```sv
parameter int TxWidth = 4;
typedef logic [TxWidth-1:0] lc_tx_t;
typedef enum lc_tx_t {
  On  = 4'b1010,
  Off = 4'b0101
} lc_tx_e;
```
In the source file `prim_lc_sync.sv` this type is used as follows:
```sv
prim_flop_2sync #(
  .Width(lc_ctrl_pkg::TxWidth),
  .ResetValue(lc_ctrl_pkg::TxWidth'(lc_ctrl_pkg::Off))
) u_prim_flop_2sync (
  ...
);
```

What sv2v generates is the following:
```sv
localparam signed [31:0] lc_ctrl_pkg_TxWidth = 4;
prim_flop_2sync #(
  .Width(lc_ctrl_pkg_TxWidth),
  .ResetValue(32'sd4'(lc_ctrl_pkg_Off))
) u_prim_flop_2sync(
 ...
);
```
meaning, it doesn't define `lc_ctrl_pkg_Off`.

However, if I change the enum definition in the package to
```sv
parameter int TxWidth = 4;
typedef logic [TxWidth-1:0] lc_tx_t;
typedef enum logic [TxWidth-1:0] {
  On  = 4'b1010,
  Off = 4'b0101
} lc_tx_e;
```
The tool produces
```sv
localparam signed [31:0] lc_ctrl_pkg_TxWidth = 4;
localparam [3:0] lc_ctrl_pkg_Off = 4'b0101;
prim_flop_2sync #(
  .Width(lc_ctrl_pkg_TxWidth),
  .ResetValue(sv2v_cast_4(lc_ctrl_pkg_Off))
) u_prim_flop_2sync(
 ...
);
```
as expected.

Please let me know if you need further information. So far I didn't check if the problem also occurs if the enum definition is in the same source file instead of a package. If it helps, I should also be able to send you over some files to reproduce this.

zachjs, 2021-04-09:
Thank you for filing this detailed issue! Indeed, the package conversion was not resolving typenames used as or within the base type of an enum. This should not be resolved on master. Please let me know if you encounter other issues!

vogelpi, 2021-04-13:
Hey @zachjs ,
thanks for the quick response and the fix!

Unfortunately, the problem still persists. To reproduce it, you can change the newly created testcase `package_enum_5.sv` as follows:
```sv
module top;
   logic [P::W-1:0] test;
   assign test = P::A;
    //initial $display("%d %d %d %b %b", $bits(P::A), $bits(P::T), $bits(P::E), P::A, P::B);
endmodule
```
This results in:
```sv
module top;
	localparam signed [31:0] P_W = 4;
	wire [3:0] test;
	assign test = P_A;
endmodule
```
If you change the definition of `test` from `logic [P::W-1:0]` to `P::E` it works. One could argue that defining test as `logic` doesn't make sense in the first place. But as in the original example, the same can happen if assigning a parameter of a module. By the way, the problem persists when adding a cast in the assignment, i.e., `assign test = P::T'(P::A);`.

Best regards

zachjs, 2021-04-13:
Thank you for the follow up! There was indeed a separate issue where the information defining the enum items could be lost if the enum type was not explicitly used in the module and the enum's base type was not a primitive. This should now be fixed as of 4ddbff9b9793e0d9d7110eddb4868e5e3f724be5. Please let me know if it's working now!

zachjs, 2021-04-13:
I didn't intend to close this. Please close it once you have confirmed the fix!

vogelpi, 2021-04-14:
It's working now, thanks a lot @zachjs for the awesome work!

closed 2021-04-14 by vogelpi