zachjs/sv2v #135: Parser fail: integer sized by localparam

chrivers, 2021-04-10:
I noticed this other construction that sv2v fails to parse. It might a little bit in a grey area, validity-wise, but yosys groks it just fine.

See the test case below:

```verilog
module foo();

    localparam BLOCK_SIZE = 1024;

    /* works with: sv2v, yosys, iverilog */
    parameter WORKS = {BLOCK_SIZE{1'bx}};

    /* works with: yosys. fails with: sv2v, iverilog */
    parameter FAILS = BLOCK_SIZE'bx;

endmodule
```

zachjs, 2021-04-10:
I'm not sure what basis in the LRM that behavior in Yosys has. Are you aware of any other tools which allow that? The closest legal thing I'm aware of would be a proper size cast: `BLOCK_SIZE'('bx)`. If `BLOCK_SIZE` were a macro, then of course something like `` `BLOCK_SIZE'bx`` would work.

chrivers, 2021-04-10:
I'm not aware either - I just discovered a difference. I suppose yosys' implementation is not overly preoccupied with being strictly standards-compliant.

I also just noticed that yosys officially recommends *not* trusting yosys' decisions on language validity. Which is.. surprising.

But all in all, no real bug here. Feel free to close.

zachjs, 2021-04-12:
Given the above discussion, I do not intend to add support for this non-standards-compliant behavior.

closed 2021-04-12 by zachjs