zachjs/sv2v #136: Ergonomics improvement: trailing commas [feature request]

chrivers, 2021-04-10:
There seems to be no good consensus on strictness vs ergonomics when it comes to trailing commas in parameter lists.

For example, this test case:

```verilog
module foo #(
    parameter THING = 0,
)();

endmodule
```

This is rejected by iverilog and sv2v, but happily accepted by yosys.

For those using sv2v as a frontend to (for example) yosys, it would be great if sv2v could accept trailing commas like this, and just not output them. That would make a good workflow for humans, and be fully compatible with the next step in the pipeline.

zachjs, 2021-04-23:
Sorry for the delay in addressing this! I've added support as of 5a8801a45f4ebe729ea88a85f38425d04799acf9. Please let me know what you think!

chrivers, 2021-05-09:
Seems to work great - thank you!

closed 2021-05-09 by chrivers