zachjs/sv2v #137: sv2v generates logic in `-E logic` mode

chrivers, 2021-04-10:
When running sv2v over the following input:

```verilog
module foo();

    task demo(input signal);
    endtask

endmodule
```

`sv2v -E logic task-logic.v`

gives

```verilog
module foo;
        task demo;
                input logic signal;
                ;
        endtask
endmodule
```

Leading to a yosys compilation error when not using `read_verilog -sv`. I think this is a bug?

zachjs, 2021-04-10:
Using `-E logic` means "don't remove the `logic` keyword or replace it with `reg` or `wire`. Excluding conversions is meant to leave certain constructs in their SystemVerilog form because the downstream tool already supports them, presumably _in SystemVerilog mode_. It seems to me the behavior here both in sv2v and in Yosys is as-intended. Perhaps the documentation is unclear?

chrivers, 2021-04-10:
Okay, that makes sense now.

I'm new to sv/v programming, but if I understand correctly, `logic` is a systemverilog-only keyword?

The surprising part, for me, is that sv2v puts `logic` into the output, when it's *not* present in the input. I guess as long as the next stage can accept it, everything should still work.

So yeah, maybe also not a bug :)

zachjs, 2021-04-11:
> `logic` is a systemverilog-only keyword?

That's correct.

> The surprising part, for me, is that sv2v puts `logic` into the output, when it's _not_ present in the input.

I agree that behavior was surprising. I've removed it as I don't believe it's necessary.

closed 2021-04-14 by zachjs