zachjs/sv2v #138: usability problem: sv2v will accept any identifier name

chrivers, 2021-04-10:
I've been looking into using sv2v to add interface/modport support to an existing verilog project. The workflow entails preprocessing the code with sv2v, and feed the resulting combined file into yosys.

The overall process works. I've managed to take the existing verilog code, and generate what is (essentially) a concatenated output file. This file still works when compiled with yosys.

After making a fairly trivial change (grouping some non-critical gpio wires into an interface + using modport), the project still compiles, but no longer works at all. There is zero signs of life from the fpga with this bitstream. I have tried compiling with/without these changes, and there's no doubt the (supposedly) trivial gpio cleanup is breaking the build.

Now, both sv2v and yosys seem quite happy with the input (no warnings from either), so I've been trying to isolate a test case for this, but since there's no indication where the problem is, this is a bit tricky in practice.

However, I did manage to find something surprising. Consider the following code:

```verilog
`default_nettype none

/* let's say the real gpio interface looks something like this */
interface gpio_port #(WIDTH=32)();
    reg [WIDTH-1:0] e;
    reg [WIDTH-1:0] o;
    reg [WIDTH-1:0] i;

    modport master(output e, output o, input i);
    modport slave(input e, input o, output i);
endinterface

/* pretend this is doing something productive */
module Blue(input blue_clk, gpio_port.master blue_gpio);
    always @ (posedge blue_clk) begin
        blue_gpio.i[0] <= red_gpio.o[0] + red_clk;
    end
endmodule

/* some middleware module, which uses Blue */
module Green (input green_clk, gpio_port.master green_gpio);
    Blue blue_gpio (.blue_clk(green_clk), .blue_gpio(green_gpio));
endmodule

/* our top module, which uses Green (which in turn uses Blue) */
module Red (input red_clk);
    gpio_port red_gpio();

    Green green (.green_clk(red_clk), .green_gpio(red_gpio.master));
endmodule
```

Now, the Blue module has no name in scope called either `red_gpio`, nor `red_clk`, but can quite happily accept both. In fact, sv2v will accept *any* name, which I guess can make sense in certain circumstances (perhaps for files meant to be included or concatenated?)

However, this *will* mean that a major amount of Hidden Magic happens. If a typo or other naming error happens to clash with almost any existing name, sv2v will quite happily generate code that does something very different than expected.

I can see certain use cases for this, but at the same time, it's a truly enormous footgun.

Perhaps it would make more sense to require identifiers (at least ones without a prefix) to be valid in the local scope?

As I said, this might not be a bug per se, but it makes the behavior of the resulting code so hard to predict, that it's almost impossible to track down why my design breaks.

Suggestions very welcome.

zachjs, 2021-04-12:
I originally imagined sv2v would be used as part of an independent FOSS-only flow, with most of the development cycle spent using a commercial suite. While this usage is indeed common, there seems to be a fair number of folks using sv2v as the first step in _all_ their flows, especially given the inconsistent SystemVerilog support existing across different commercial suites. As such, catching/handling invalid input was not a major focus. While I don't think it's necessary for sv2v to contain a proper type checker, I do think there would be significant value in checking _more_. This will be a significant undertaking because the existing architecture doesn't accommodate producing useful error messages with source information. I don't have a concrete plan at this point, but know that I am looking into it.

chrivers, 2021-04-15:
Just to make sure I'm not misunderstanding you - I'm *only* using (or trying to use) sv2v in a yosys workflow. I don't have any commercial tools available.

I understand your argument about relying on the backend-tool for type checking (etc), and I agree.

However, this is both a simpler, and, in a sense, worse issue.

Even without paying *any* attention to types, sv2v generates code that can link wires/nets that have no relation.

This is purely a name resolution problem.

For example, in the snippet above, the innermost module, `Blue`, refers to a name that is not in its local scope, `red_clk`. If we completely disregard typing, my limited understanding of verilog gives me the impression that there are only 2 reasonable outcomes: 
  1) The name `red_clk` refers to a valid name, that isn't visible from the current scope. This could happen if the file is generated to be included in a larger file, for example.
  2) The name `red_clk` is invalid, and should generate a reasonable error when compiled.

However, what actually happens now, is:

  3) The name `red_clk` happens to clash with a name in an outer module (`Red`), and the generated code will connect these wires.

This seems wrong to me, because
  A) It hides the error, and
  B) It *breaks* workflow 1 above, which might have been valid otherwise.

Again, I don't see the need to do any type checking here (but I'm new to verilog, so I might be wrong). To me, this seems like purely a problem with name resolution.

I think sv2v already does some amount of name resolution. If the generated output names were unique-ified, this problem would not happen.

As of right now, I've been trying to use sv2v for my yosys workflow, but the sv2v conversion breaks as soon as I add an interface modport, and because the generated code is suddenly considerably different, I'm finding it hard to impossible to debug what's going wrong.

I *think* it's related to this issue.

zachjs, 2021-04-18:
I completely agree that sv2v taking this _invalid_ SystemVerilog and silently producing _valid_ Verilog represents a substantial usability problem. The second outcome you've described is what I intend to happen: sv2v should catch this issue and raise an error. While I'm not committed to this strategy or implementation, I've made a branch called [`interface-check-expr-resolution`](https://github.com/zachjs/sv2v/tree/interface-check-expr-resolution) which attempts to ensure that the inlining performed in the interface conversion will not allow names to resolve that would not resolve otherwise. When run on your example above it raises an error: `` inlining instance "blue_gpio" of module "Blue" would make expression "red_gpio.o" used in "blue_gpio" resolvable when it wasn't previously ``. Please let me know what you think!

While I agree the first outcome (leaving it to resolve to some other global) would potentially be reasonable, this is not intended usage. sv2v generally needs access to all of the relevant sources to be able to convert things correctly. I don't think it's feasible to convert sources "piecewise" unless the relevant information can be represented in equivalent Verilog. For example, sv2v does not allow the independent "conversion" of packages in one invocation and conversion of the modules that use them in a separate invocation.

In my (perhaps naive) assessment, properly checking the name resolution of the source prior to the downstream conversions mangling things as shown in this issue is tantamount to resolving the types of expressions "before" compilation, which is most of the work in creating type checker. I agree a type checker is not necessary, and simply used that terminology as a shorthand for the kind of name resolution it entails.

The changes I've made on that branch are obviously far simpler than fully checking the source for validity. My hope is that any invalid input will be caught _either_ in sv2v _or_ in the downstream tool. Certainly the more checks sv2v can do, the better.

chrivers, 2021-05-09:
Hello again - and sorry for the delay.

I've tried the new branch in various ways, and it seems like a big improvement! At least one major test built into a broken system earlier, and now works.

I'm not sure if that's due to any logic in the new branch, or because of other changes in the meantime. But something improved, for sure :+1: 

Now, there are some rough edges. Consider this code:

```verilog
interface ValidIntf;
    
endinterface

module mod();
    ValidIntf intf();

    thing foo (
        .validname(intf)
    );
endmodule

module thing(
    ValidIntf.BadName validname
);

endmodule
```

Running this through sv2v produces only the following output:

```
sv2v: src/Convert/Interface.hs:(231,35)-(232,70): Non-exhaustive patterns in Just (accesses, _, _)
```

Now, the input code isn't valid, so I think it's mostly a matter of error message ergonomics.

I hit this error due to a typo, and started looking for a minimal reproducer.

Comments?

zachjs, 2021-05-13:
I have merged the expression resolution checking into master. I've also just pushed some work to improve the checking of the types of modports (module definition vs. binding). Please let me know what you think!

zachjs, 2021-07-06:
Have you had a chance to try out these improvements?

zachjs, 2021-07-30:
Given the issues have been addressed and the recent inactivity, I am closing this issue. Please feel free to reopen or file a new one if you find something!

closed 2021-07-30 by zachjs