zachjs/sv2v #140: Multiple issues related to structures and

flaviens, 2021-04-13:
Hello!

Thank you for your work! sv2v is a great tool!

I have noticed that some structures remain after a sv2v pass, here is a way to reproduce them on a real open source design:

0. (prerequisite) Make sure to have [bender](https://github.com/pulp-platform/bender) installed:
```
cargo install bender
```

1. Clone a dedicated fork originated from openhwgroup/cva6 and update the submodules:
```
git clone https://github.com/flaviens/cva6.git
cd cva6
git submodule update --init --recursive
git checkout sv2v_issues
```

2. Run sv2v on a selected file set, include directory and defines:
```
make -C sv2v
```

3. Replace some dependency:
```
sed -i 's/synopsys_sram.sv/util\/sram.sv/g' Bender.yml
```

4. Observe the output located in `sv2v/sv2v_out.v`. The output is mainly correct but some issues remain, which are reveled by using Yosys with the command `yosys read_verilog -sv sv2v/sv2v_out.v`.

Notably:
- **Lines 496, 504 and 529**: an occurrence of `fopen` `string` and `final` remains.
- **Line 25181**: `rdata_i[j * 174+:174].tag` and `rdata_i[j * 174+:174].valid` appear in the output, which is apparently not Verilog, since structures are not supported. Surprisingly, this is the only line in the output file where the substring `].` occur.
- **Lines 387823, 38847, 38972 and 38993**: parameters are kept in SystemVerilog form, for example
```verilog
.DataType(struct packed {
    logic [stream_arbiter_flushable_ED0DD.DATA_T_DATA_T_AUX_WIDTH - 1:0] aux;
    logic [stream_arbiter_flushable_ED0DD.DATA_T_DATA_T_ID_WIDTH - 1:0] id;
}),
```
- **Line 41486**: `parameter type DataType = logic [DataWidth - 1:0];`.

Thank you again for maintaining sv2v. I hope that this post will help you improve it. Also, please let me know if I did something wrong.

zachjs, 2021-04-14:
Thanks for providing these instructions. This conversion demonstrates issues even beyond those you've mentioned here. I am working on getting these issues fixed up. However, I do have a few comments regarding your first note.

`$fopen`, etc., are valid Verilog, so I'm not sure what transformations are expected here. If the downstream use is for synthesis, I would imagine those blocks should be ifdef'd away. If the downstream tool is a simulator, arguably the simulator needs to have support for those system tasks.

`final` has no Verilog equivalent I'm aware of. This is another simulation-only construct. What would you suggest here?

The way the module uses `string` is a bit odd. It seems like `byte` would work instead. sv2v does already handle string parameters in way that allows for the string parameter to be variable-length, but this strategy doesn't work for procedural code. Arguably static analysis could arrive at some maximum length for _some_ procedural uses of `string`, but even that seems quite convoluted. Do you have an idea of what strategy might make sense here?

flaviens, 2021-04-14:
Hi Zachjs,

Thank you for your fast reaction, that's great to read :slightly_smiling_face: !
- The `$fopen` was indeed a mistake of mine, induced by the `final` keyword of `$fclose`.
- There is no single answer I think and I believe you would propose better answers than me :slightly_smiling_face: A naive possibility is to identify all the `final` blocks and run them just before each occurrence of `$finish` in the code. But you may propose a much more elegant answer.
- Again, I do not know for sure which is the better solution. This could be replaced by a `byte` in this code indeed, but this would naturally not provide a general answer.

zachjs, 2021-04-15:
I've addressed the issues I'm aware of in the output, aside from the instances of `final` and `string` we are discussing. Please let know what you find!

Can you clarify for me what downstream tool(s) you're targeting? It seems we can rule out synthesis, as the script invoking sv2v here defines the `VERILATOR` flag to enable certain simulation-only constructs.

If you intend to use the result in simulation, what simulator are you using? I'm not personally aware of a simulator which _does_ support `$fopen`, but _doesn't_ support `final` and `string`. I ask this because if we need to find some alternative for `final` and `string`, it would be very helpful to know what constructs the target supports.

flaviens, 2021-04-16:
Hi Zachjs,

Thank you again for your reactivity! And indeed you did a great job, many issues are solved!

- The direct downstream target tool is Yosys, but whose output could be simulation (Verilator) or further synthesis / P&R.
- I indeed set the `Verilator` flag but that is not cast in stone. It was mostly because of the dependency on some `common_cells`.

**There are still some incompatibilities with Yosys**, that you can explore using the same branch that I just updated for you (besides, I commented out the problematic code with the string and final in `ariane.sv` in this fork).

I also needed to comment out the `find_first_one` module because one of its parameters remains -1 and this results in a huge loop in Yosys's parsing step. But I am not convinced that it is related to any potential issue in sv2v at all.

Please let me know if you need any additional information!

zachjs, 2021-04-17:
I pulled your branch, and with one small modification to the Yosys script, it progresses much further. There are some warnings and a failure in `PROC_DFF` that I have not yet had the chance to look into.

The latest Yosys release (0.9) is well behind master. You should build the latest Yosys as it contains a number of fixes and improvements.

I recommend passing the `-defer` flag to `read_verilog`, which tells Yosys not to attempt to elaborate modules using their default parameters. This will fix the issue with `find_first_one`. The modules are elaborated as needed as part of the `hierarchy` pass.

flaviens, 2021-04-19:
Thank you very much for your fixes and tips, your work is extremely valuable.

closed 2021-04-19 by flaviens