zachjs/sv2v #141: Converting dimensions from unpacked to packed if connected to port

vogelpi, 2021-04-16:
Hi Zach,
I am playing around with sv2v and Yosys and noted some unexpected behavior also observable in the following simplified snippet:
```sv
package P;
  localparam int N1 = 4;
  localparam int N2 = 2; 

  localparam int W = 3;
  typedef enum logic [W-1:0] {
    A = 3'b011,
    B = 3'b100
  } E;
endpackage

module child import P::*;
(
  input  logic [N1*N2-1:0] in_i,
  output E        [N1-1:0] out_o [N2]
);

  for (genvar n2 = 0; n2 < N2; n2++) begin
    for (genvar n1 = 0; n1 < N1; n1++) begin
      assign out_o[n2][n1] = (in_i == 1'b1) ? A : B;
    end
  end

endmodule

module top import P::*; (
  input  logic [N1*N2-1:0] in_i,
  output logic [N1*N2-1:0] out_o
);

  E [N1-1:0] sig1 [N2];
  E [N1-1:0] sig2 [N2];

  child u_child (
    .in_i,
    .out_o (sig2)
  );

  for (genvar n2 = 0; n2 < N2; n2++) begin
    for (genvar n1 = 0; n1 < N1; n1++) begin
      assign sig1[n2][n1] = sig2[n2][n1];
    end
  end

  for (genvar n2 = 0; n2 < N2; n2++) begin
    for (genvar n1 = 0; n1 < N1; n1++) begin
      assign out_o[n2*N1+n1] = (sig1[n2][n1] == A) ? 1'b1 : 1'b0;
    end
  end
endmodule
```

Basically, I have two unpacked arrays `sig1` and `sig2` that I connect together. On of them is connected to a module port. Now, since Verilog doesn't know unpacked array ports, `sv2v` converts the port to a packed array. Along with the port, also `sig2` is converted. This results in the following Verilog code:
```verilog
module child (
	in_i,
	out_o
);
	localparam signed [31:0] P_N1 = 4;
	localparam signed [31:0] P_N2 = 2;
	input wire [7:0] in_i;
	localparam signed [31:0] P_W = 3;
	output wire [23:0] out_o;
	localparam [2:0] P_A = 3'b011;
	localparam [2:0] P_B = 3'b100;
	generate
		genvar n2;
		for (n2 = 0; n2 < P_N2; n2 = n2 + 1) begin
			genvar n1;
			for (n1 = 0; n1 < P_N1; n1 = n1 + 1) assign out_o[(((1 - n2) * P_N1) + n1) * 3+:3] = (in_i == 1'b1 ? P_A : P_B);
		end
	endgenerate
endmodule
module top (
	in_i,
	out_o
);
	localparam signed [31:0] P_N1 = 4;
	localparam signed [31:0] P_N2 = 2;
	input wire [7:0] in_i;
	output wire [7:0] out_o;
	localparam signed [31:0] P_W = 3;
	wire [11:0] sig1 [0:1];
	wire [23:0] sig2;
	child u_child(
		.in_i(in_i),
		.out_o(sig2)
	);
	generate
		genvar n2;
		for (n2 = 0; n2 < P_N2; n2 = n2 + 1) begin
			genvar n1;
			for (n1 = 0; n1 < P_N1; n1 = n1 + 1) assign sig1[n2][n1] = sig2[(((1 - n2) * P_N1) + n1) * 3+:3];
		end
	endgenerate
	localparam [2:0] P_A = 3'b011;
	generate
		for (n2 = 0; n2 < P_N2; n2 = n2 + 1) begin
			genvar n1;
			for (n1 = 0; n1 < P_N1; n1 = n1 + 1) assign out_o[(n2 * P_N1) + n1] = (sig1[n2][n1] == 3'b011 ? 1'b1 : 1'b0);
		end
	endgenerate
endmodule
```
I think this is code is correct. However, synthesizing this with Yosys causes `sig1` to be optimized away completely. `out_o` is constantly `8'h00`.

If I just change the assignment of `sig1` from the loop to
```sv
  assign sig1 = sig2;
```
then both `sig1` and `sig2` are defined as `wire [23:0]`. The resulting code can be correctly synthesized. I am not sure if this is a bug in Yosys or if `sv2v` should always convert both signals to packed arrays anyway. Do you have an idea?

Best,
Pirmin

zachjs, 2021-04-16:
Thank you for filing this issue! There was indeed a bug in sv2v, which I have now resolved in 499bd5873ee25b118de2d577972af0ce262340f6. Please let me know if this works for you!

In the prior output, `` assign sig1[n2][n1] = sig2[(((1 - n2) * P_N1) + n1) * 3+:3]; `` was incorrect. The two packed dimensions of `sig1` are flattened, so the left hand side should be accessing chunks of size 3.

Before the fix, the "multipack" (flattening) conversion was considering:
* type: `sig1`, access: `[n2][n1]`
* type: `sig1`, access: `[n2]`

But because the most significant dimensions was unpacked, it really needed to consider:
* type: `sig1[n2]`, access: `[n1]`

A similar issue existed with part-selects/slices which has also been resolved.

You also noted that `sig1` is not fully flattened. In an effort to preserve memories, sv2v leaves single unpacked dimensions as-is unless they "must" be made packed and flattened. This is true of `sig2` because it is bound to a port. While it's possible leaving `sig2` partially un-flattened would be legal in Verilog-2005, I know that Yosys does not support this, so the dimension is packed and flattened.

vogelpi, 2021-04-19:
Ah, this makes sense. The problem in the LHS of the `sig1` assignment is pretty obvious now. I guess I didn't see anymore the wood for the trees.

It's working now. Thanks for the quick fix and the explanation! I really appreciate it.

closed 2021-04-19 by vogelpi