zachjs/sv2v #142: sv2v takes too much RAM space and gets killed by the OS

iamrishabh23, 2021-04-30:
Hi ,
I have a few files which I was converting and I observed this error message from the OS stating that the task was killed
[image]

Will be attaching the files which I have used .
[TestFiles.zip](https://github.com/zachjs/sv2v/files/6407275/TestFiles.zip)
 Command used :
 
sv2v Main_Zynq_Wrapper.sv Main_Zynq.sv Core.sv NextPCStage.sv NextPCStageIF.sv

Also observed through system monitor that the Memory usage maxes out and gets killed
OS: Ubuntu 20.04
RAM:16GB

zachjs, 2021-04-30:
I cannot yet reproduce the issue yet because the file `XilinxMacros.vh` is missing. Please double check your command and ZIP.

iamrishabh23, 2021-05-01:
Hi Zach ,
I observe it doesn't complain about XilinxMacros.vh  for me . The command I used was the same as above the executable from bin file.
i.e >>  ~/Tools/sv2v/bin/sv2v  --write=adjacent Main_Zynq_Wrapper.sv Main_Zynq.sv Core.sv NextPCStage.sv NextPCStageIF.sv
OR
    >>~/Tools/sv2v/bin/sv2v   Main_Zynq_Wrapper.sv Main_Zynq.sv Core.sv NextPCStage.sv NextPCStageIF.sv

Adding the file XilinxMacros.vh 
[Files_w_XilinxMacros_RAM_Issue.zip](https://github.com/zachjs/sv2v/files/6409167/Files_w_XilinxMacros_RAM_Issue.zip)

iamrishabh23, 2021-05-01:
Hi Zach , 
Adding to this , I think there is an issue with my setup and because of which just trying to convert  one file is also causing the process to be killed. 
I don't think it is because of the files used. How do I capture this issue for you to reproduce ?

iamrishabh23, 2021-05-01:
i ran the following command to check processes which take memory before getting killed
dmesg -T| grep -E -i -B100 'killed process'

THe last few processes ran as as follows:

[Sat May  1 04:04:28 2021] [  12267]  1000 12267    10996      157    77824        0             0 obexd
[Sat May  1 04:04:28 2021] [  12302]   123 12302    39056      150    65536        0             0 dconf-service
[Sat May  1 04:04:28 2021] [  12349]  1000 12349  9839360    18247  1425408        0           300 chrome
[Sat May  1 04:04:28 2021] [  13208]  1000 13208  9771041     3272   942080        0           300 chrome
[Sat May  1 04:04:28 2021] [  13328]  1000 13328   168458     3633   245760        0             0 gnome-system-mo
[Sat May  1 04:04:28 2021] [  13406]  1000 13406     2841      488    69632        0             0 bash
[Sat May  1 04:04:28 2021] [  13587]  1000 13587   188497     3701   278528        0             0 gvim
**[Sat May  1 04:04:28 2021] [  14700]  1000 14700 268903321  2480416 20275200        0             0 sv2v**

[Sat May  1 04:04:28 2021] oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),cpuset=/,mems_allowed=0,global_oom,task_memcg=/user.slice/user-1000.slice/user@1000.service,task=sv2v,pid=14700,uid=1000
[Sat May  1 04:04:28 2021] Out of memory: Killed process 14700 (sv2v) total-vm:1075613284kB, anon-rss:9921664kB, file-rss:0kB, shmem-rss:0kB, UID:1000 pgtables:19800kB oom_score_adj:0

iamrishabh23, 2021-05-02:
Closing this issue as It was pointed out to be a recurring `include within the file itself . Thank you for the clarification .

closed 2021-05-02 by iamrishabh23