zachjs/sv2v #145: latest git failed to build under Ubuntu 20.04 LTS (focal)

fdarling, 2021-05-14:
I am trying to build sv2v from the latest git on Ubuntu 20.04 LTS (focal fossa) amd64. Here is some information about my system:

```
$ dpkg -l | grep git | head -n 1
ii  git              1:2.25.1-1ubuntu3.1    amd64    fast, scalable, distributed revision control system

$ dpkg -l | grep haskell-stack
ii  haskell-stack    1.9.3.1-1              amd64    The Haskell Tool Stack

$ uname -r
5.8.0-53-generic

$ lsb_release -r
Release:	20.04
```

I installed the Ubuntu package `haskell-stack`:

```
$ sudo apt-get install haskell-stack
```
Cloned the latest git:

```
$ git clone https://github.com/zachjs/sv2v.git
```

And tried building using `make`:

```
$ cd sv2v
$ make
mkdir -p bin
stack install --install-ghc --local-bin-path bin
Downloading lts-16.14 build plan ...
RedownloadInvalidResponse Request {
  host                 = "raw.githubusercontent.com"
  port                 = 443
  secure               = True
  requestHeaders       = []
  path                 = "/fpco/lts-haskell/master//lts-16.14.yaml"
  queryString          = ""
  method               = "GET"
  proxy                = Nothing
  rawBody              = False
  redirectCount        = 10
  responseTimeout      = ResponseTimeoutDefault
  requestVersion       = HTTP/1.1
}
 "/home/forest/.stack/build-plan/lts-16.14.yaml" (Response {responseStatus = Status {statusCode = 404, statusMessage = "Not Found"}, responseVersion = HTTP/1.1, responseHeaders = [("Connection","keep-alive"),("Content-Length","14"),("Content-Security-Policy","default-src 'none'; style-src 'unsafe-inline'; sandbox"),("Strict-Transport-Security","max-age=31536000"),("X-Content-Type-Options","nosniff"),("X-Frame-Options","deny"),("X-XSS-Protection","1; mode=block"),("Content-Type","text/plain; charset=utf-8"),("X-GitHub-Request-Id","3CD6:205A:597277:A49446:609E75F1"),("Accept-Ranges","bytes"),("Date","Fri, 14 May 2021 13:06:57 GMT"),("Via","1.1 varnish"),("X-Served-By","cache-pwk4930-PWK"),("X-Cache","MISS"),("X-Cache-Hits","0"),("X-Timer","S1620997618.865914,VS0,VE80"),("Vary","Authorization,Accept-Encoding"),("Access-Control-Allow-Origin","*"),("X-Fastly-Request-ID","a0b6519c1ea7d3c862c0ab8a1040f40eaa1a2b2d"),("Expires","Fri, 14 May 2021 13:11:57 GMT"),("Source-Age","0")], responseBody = (), responseCookieJar = CJ {expose = []}, responseClose' = ResponseClose})
make: *** [Makefile:7: sv2v] Error 1
```

However, it seems that build process is having trouble downloading a necessary file. I am newbie to sv2v, so any help would be appreciated!

zachjs, 2021-05-14:
If `1.9.3.1` is your Stack version, I recommend upgrading your version. Based on the error message, it appears to be using a stale repo. I am currently on `2.5.1.1`, and the latest is `2.7.1`.

zachjs, 2021-05-21:
Any luck with upgrading your stack version? If this is an issue with sv2v, I'm eager to get it fixed!

fdarling, 2021-05-22:
It was definitely the version of stack. I downloaded a newer version (version 2.7.1) and set it up locally in my home folder:

```
$ ./stack --version
Version 2.7.1, Git revision 8afe0c2932716b0441cf4440d6942c59568b6b19 x86_64 hpack-0.34.4
```

Using that version of `stack`, I was able to build `sv2v`. It also ran without any problems, so you can mark this issue as closed with a solution. Thanks!

closed 2021-05-22 by zachjs