zachjs/sv2v #146: Some module syntax is not accepted

flaviens, 2021-05-19:
Hi!

I've encountered some Verilog file that could not be parsed by sv2v: aes_128.sv (line 32) of AES_T100 (and probably others as well).
You can find the file here: https://www.trust-hub.org/#/benchmarks/chip-level-trojan.

It seems to be due to an unusual sequence of module instantiations separated by a comma.

zachjs, 2021-05-21:
I've added support for unconnected ports in ordered port connections as of 801955ffab6ce4d89e1a274aeb34ffce6a771e21. This is done in a way which I believe is logically consistent with the support for trailing commas added previously.

I've confirmed the reference file now converts cleanly. Please let me know if this works for your use case!

flaviens, 2021-05-21:
Great, perfect! Thank you for your work!

closed 2021-05-21 by flaviens