zachjs/sv2v #148: Potential issue with passing type parameter into an Interface instantiation.

andrewandrepowell, 2021-05-29:
Hello!

I'm running into the following issue when I try to instantiate an interface with a parameter type.

```bash
v2v sfifo.sv 
sv2v: interface param T expected type, found expr: T
CallStack (from HasCallStack):
  error, called at src/Convert/Interface.hs:561:36 in main:Convert.Interface
```

It appears if I directly pass the type into the instantiation, it works fine. For instance, `flow #(.T(T)) send();` fails but `flow #(.T(logic[3:0])) send();` works without any problems. 

Here's the full system verilog code.

```systemverilog

interface flow (valid, data);
  parameter type T;
  logic valid;
  T data;
  modport sender(output data, output valid);
  modport receiver(input data, input valid);
endinterface;

module stage(clock, reset, receive, send);
  input logic clock, reset;
  flow.receiver receive;
  flow.sender send;

  always @(posedge clock)
  begin
    send.data <= receive.data;
    send.valid <= receive.valid;
  end
endmodule;

module stage_imp(clock, reset, send_data, send_valid, receive_data, receive_valid);
  localparam type T = logic[15:0];
  input logic clock, reset;
  input T receive_data;
  input logic receive_valid;
  output T send_data;
  output logic send_valid;

  assign receive.valid = receive_valid;
  assign receive.data = receive_data;
  assign send_data = send.data;
  assign send_valid = send.valid;

  flow #(.T(T)) send();
  flow #(.T(T)) receive();
  stage dut(.clock(clock), .reset(reset), .receive(receive), .send(send));

endmodule;
```

zachjs, 2021-05-31:
Thanks for filing this issue! This actually uncovered several closely related problems too. Potential ambiguities between types and expressions should now be resolved on master. Please let me know if it works for you!

andrewandrepowell, 2021-06-02:
Works like a charm! Appreciate the prompt response.

closed 2021-06-02 by andrewandrepowell