zachjs/sv2v #149: Not conveting struct packed

zeeshanrafique23, 2021-06-03:
I am getting the same struct packed after conversion, and sometimes it is not converting even the 'logic' variable inside structure, and also it is not supporting input parameter "tag" type.
Having issues while converting this module. [rr_arb_tree.sv](https://github.com/merledu/azadi_apr/blob/8f9580a7192bc1922c956fc2b7681eb822c2f876/azadi_rtl_v0.6/rr_arb_tree.sv#L53)

zeeshanrafique23, 2021-06-03:
It would be great if someone can do it, on priority basis. @zachjs

zachjs, 2021-06-03:
Can you share full reproduction steps? When I run `sv2v -DVERILATOR rr_arb_tree.sv`, it seems to convert without obvious issues. It seems likely that the specific instantiations are important here. I'm also not sure what you mean by not "supporting input parameter 'tag' type."

zeeshanrafique23, 2021-06-03:
Well here are my all [files](https://github.com/merledu/azadi_apr/tree/main/azadi_rtl_v0.6) which I want to convert, and [this repo](https://github.com/zeeshanrafique23/azadi-netlist) contains the files which I converted today, also @olofk tried conversion by adding a python script [here](https://gist.github.com/olofk/cc8f16702ef3ed5261660f57ee050c16), but there is still problem with the rr_arb_tree.sv module.

You can also check this [issue](https://github.com/alainmarcel/UHDM/issues/426) where I mentioned the issue in detail.

zachjs, 2021-06-03:
Based on the Python script, it looks like you are converting each file separately. In general, sv2v must be invoked with all of the relevant source files simultaneously in order to perform the conversions.

zeeshanrafique23, 2021-06-03:
Yes I did it in this way, I put all the sv modules and packages in one file and performed sv2v in this [repo](https://github.com/zeeshanrafique23/azadi-netlist)

zeeshanrafique23, 2021-06-03:
@zachjs can you do the conversion on your end for the files (I'll post link shortly), because the time is short I have to submit this design in OpenMPW-TWO and the yosys is generating errors messages on the files which I converted.

zachjs, 2021-06-04:
I've updated master with fixes for a few different issues I encountered converting the linked project. It is now ingested by Yosys without error, but there are some warnings which may or may not be innocuous. Please let me know if it works!

zeeshanrafique23, 2021-06-04:
Great thanks I’ll let you know

zeeshanrafique23, 2021-06-04:
That error was resolved on yosys but getting error while flattening the design 
```
<suppressed ~440 debug messages>
ERROR: Mismatch in directionality for cell port $paramod$97cb237d24228be5f7c4821454f589d09968902f\fpnew_opgroup_multifmt_slice_7C482.gen_num_lanes[0].active_lane.lane_instance.i_fpnew_cast_multi.extension_bit_o: \lane_ext_bit [0] <= \gen_num_lanes[0].active_lane.lane_instance.i_fpnew_cast_multi.extension_bit_o
```
I am trying to understand this error deeply may be this is a problem related to RTL designing.

zeeshanrafique23, 2021-06-04:
@zachjs did you synthesize the all files? Or just rr_arb_tree

zachjs, 2021-06-04:
I passed the entire converted output into Yosys and ran hierarchy:
```
yosys -p "read_verilog -defer -mem2reg converted.v; hierarchy -top azadi_soc_top"
```
Based on that error message, the version of Yosys you are using is not be up to date. You may have better luck using Yosys built from master, which has many changes since the latest release.

zeeshanrafique23, 2021-06-04:
Oh yes, I am using older version, thanks for pointing out the issue let me check it on latest version.

zeeshanrafique23, 2021-06-05:
Thanks man for the quick response, the issues are resolved.

closed 2021-06-05 by zeeshanrafique23