zachjs/sv2v #150: Problems about package

peijunh, 2021-06-13:
Hello, it seems that sv2v cannot deal with packages of function. Is there any way to avoid this? Below is the issue

$ sv2v --write=adjacent prim_util_pkg.sv
Warning: Source includes packages but no modules. Please convert packages alongside the modules that use them.

And the verilog is as follows:

package prim_util_pkg;
  function automatic integer _clog2(integer value);
    integer result;
    integer v = value;
    v = v - 1;
    for (result = 0; v > 0; result++) begin
      v = v >> 1;
    end
    return result;
  endfunction
  function automatic integer vbits(integer value);
`ifdef XCELIUM
    return (value == 1) ? 1 : _clog2(value);
`else
    return (value == 1) ? 1 : $clog2(value);
`endif
  endfunction
endpackage

zachjs, 2021-06-13:
Perhaps the error message is not clear. If you have files which use a particular package, that package and the files that use it must be converted simultaneously as part of a single invocation of sv2v. Would adding "simultaneously" to that error message make it more clear?

peijunh, 2021-06-13:
Thank you deeply for your reply! I think you want to say that, sv2v cannot transform a pure package, it needs to expand the package where there is a module that uses it. So this is what I am doing right now. I transform a system verilog that use a package defined in another file, and get the error message:

sv2v: could not find package "tlul_pkg"
CallStack (from HasCallStack):
  error, called at src/Convert/Package.hs:526:42 in main:Convert.Package

I run the command like this:

 sv2v --incdir=. --write=adjacent opentitan_soc_top.sv

The definition of this package is exactly in the same directory as the file I want to transform, so I quite don't understand why it does not work?

zachjs, 2021-06-13:
The include directories are only used to search for files included via the include preprocessor directive. sv2v automatically uses the directory containing each source file as a directory to search for included files. Packages are often passed alongside, rather than being included, though this varies by project.

The file list should include both the files defining the packages and the files defining the modules that use those packages, at the same time. Often this means that a single invocation of sv2v will list all of a project's source files. To be clear, your invocation should probably look something like: `sv2v fileThatDefinesPaclage.sv someFileThatUsesThePackage.sv someOtherFileThatUsesThePackage.sv`. Often globbing patterns are used.

peijunh, 2021-06-15:
Hi, I want to thank you for your help. 
Also, I want to share with you an interesting bug of my project. When I transform the system verilog to verilog, the result cannot be compiled by yosys. 

Then I debug my system verilog and found this is because I use a none constant value inside $clog, which is '$clog2( non-constant expression )'. This is not supported in design compiler, and the corresponding verilog will also create an issue in yosys.

zachjs, 2021-06-16:
Indeed, Yosys does not support the use of `$clog2` on non-constant expressions. I'm not sure if you are suggesting A) sv2v should realize that the argument to `$clog2` is not constant and raise an error; or B) sv2v should be converting the `$clog2` in some way; or C) something else.

Note that there are some usages of `$clog2` that the Yosys v0.9 release does not handle, but which are allowed if you build the latest version. For example, the following works in Yosys when it did not previously:
```Verilog
module top(
    input wire signed [31:0] inp,
    output wire signed [31:0] out
);
    function automatic integer f;
        input integer x;
        input integer y;
        f = x >> $clog2(y);
    endfunction
    assign out = f(inp, 1024);
endmodule
```

zachjs, 2021-07-06:
Are you still facing an issue here? I'm eager to help if I can!

peijunh, 2021-07-06:
Up to now, there is no issue, I really appreciate your help and I admire this sv2v which helps my project a lot!

zachjs, 2021-07-06:
That's great to hear! Please feel free to open a new issue if you run into problems in the future!

closed 2021-07-06 by zachjs