zachjs/sv2v #152: Issue with parsing some generate statements

flaviens, 2021-06-17:
Hi!

I'm back for another issue :p
Some generate statements generated by sv2v seem to be not accepted by some tools.

This is an example from OpenTitan:
```verilog
// Copyright lowRISC contributors.
// Licensed under the Apache License, Version 2.0, see LICENSE for details.
// SPDX-License-Identifier: Apache-2.0

// This file is auto-generated.

// This is to prevent AscentLint warnings in the generated
// abstract prim wrapper. These warnings occur due to the .*
// use. TODO: we may want to move these inline waivers
// into a separate, generated waiver file for consistency.
//ri lint_check_off OUTPUT_NOT_DRIVEN INPUT_NOT_READ
module prim_flop_2sync #(

  parameter int               Width      = 16,
  parameter logic [Width-1:0] ResetValue = '0

) (
  input                    clk_i,
  input                    rst_ni,
  input        [Width-1:0] d_i,
  output logic [Width-1:0] q_o
);

  if (1) begin : gen_generic
    prim_generic_flop_2sync #(
      .Width(Width),
      .ResetValue(ResetValue)
    ) u_impl_generic (
      .*
    );

  end

endmodule
//ri lint_check_on OUTPUT_NOT_DRIVEN INPUT_NOT_READ
```

If we remove the `if(1)`, then [Morty](https://github.com/zarubaf/morty) complains:

```
Error: parse error
   --> src/prim_flop_2sync.sv:13:16
   |
13 |    generate begin : gen_generic
   |                ^
Error: Parse error: Some(("prim_flop_2sync.sv", 240))
```
But Verilator accepts the file.
This follows [this original issue in the Morty repo](https://github.com/zarubaf/morty/issues/27).

Many thanks!

zachjs, 2021-06-17:
To my knowledge, IEEE 1364-2001 was the last of the specifications to allow bare generate blocks, i.e., not conditioned on `if`, `for`, or `case`. Based on how the grammar was changed, this feature's removal in IEEE 1364-2005 was intentional. I still find this surprising given the standards' proclivity for backwards compatibility.

Most tools allow this pattern always. I've encountered one which restricts it to explicit generate regions. `iverilog` generates warnings on these by default.

To increase portability, I will change sv2v to produce compatible output. It will still accept these blocks, but will add `if (1)` as needed during code generation.

I would argue that other parsers should accept them as well, especially given the precedent set by other tools, but obviously some favor strict standards adherence.

zachjs, 2021-06-17:
This should be fixed as of 108852060e78768243ee916e6f9ce6181b6bf021.

flaviens, 2021-06-29:
Great thanks!

closed 2021-06-29 by flaviens