zachjs/sv2v #156: False failure with fancy port name: unknown binding "..." specified for port connections in instance ...

mkorbel1, 2021-07-08:
I'm getting an error like this:
```
sv2v: unknown binding "..." specified for port connections in instance "..." of "..."
CallStack (from HasCallStack):
  error, called at src/Convert/ResolveBindings.hs:127:9 in main:Convert.ResolveBindings
```

But I see the port name in question defined.  I can't give the exact example, but the port names have some strange characteristics that I can try to replicate:

```
module mod_w_fancy_ports (
...
  input logic [4:0] \AFancySignalName[3].Something ,
  input logic [4:0] \AFancySignalName[3].SomethingElse
);
  ...
endmodule

...
// instantiation somewhere else
mod_w_fancy_ports inst_of_fancy_module(
  ...
  .\AFancySignalName[3].SomethingElse (\BFancySignalName.SomethingElse)
);
...
```

I expect the corresponding error for this example, replacing signal names, would be (note the double \ and space at the end in the quotes, if that is helpful):
```
sv2v: unknown binding "\\AFancySignalName[3].SomethingElse " specified for port connections in instance "inst_of_fancy_module" of "mod_w_fancy_ports "
CallStack (from HasCallStack):
  error, called at src/Convert/ResolveBindings.hs:127:9 in main:Convert.ResolveBindings
```

This code parses correctly with other tools.  I included details about the port name since I don't see similar issues elsewhere and suspect the port name might have something to do with the error.

zachjs, 2021-07-08:
Thanks for filing another detailed report! Escaped identifiers were being treated as distinct depending on the whitespace which follows them, but this should be fixed as of 91e3ac0fb17.

However, sv2v does not correctly handle cases where it generates a new identifier using an escaped identifier as a base name. There is some opportunity to refactor the handling of identifiers, but it may not be at the top of my list unless you are actually running into that problem.

mkorbel1, 2021-07-08:
Thank you for the quick fix!  That resolved the issue I was facing, and now I'm back to #155, which I'll continue debugging now.

zachjs, 2023-07-02:
I'm closing this issue as the original issue is resolved. If you run into other issue with escaped identifiers, please file a new issue. Thanks!

closed 2023-07-02 by zachjs