zachjs/sv2v #158: Direct Programming Interface (DPI) support?

ChristianB12, 2021-07-21:
Hi,

I'm trying to convert SystemVerilog that uses DPI-C to Verilog so I can run it through OpenLane.
I'm assuming there won't be support for this since you'd have to parse through every C file and implement the functionality in Verilog. Unless there is some alternative I don't know about.

Figured I'd ask though since I don't have another solution

SimDRAM.sv:1:8: Parse error: unexpected token '"DPI-C"' (Lit_string)

mithro, 2021-07-21:
You might want to consider [Surelog](https://github.com/chipsalliance/Surelog) + [UHDM](https://github.com/chipsalliance/uhdm) as an option for [parsing SystemVerilog for usage with Yosys](https://antmicro.com/blog/2020/12/ibex-support-in-verilator-yosys-via-uhdm-surelog/). The team at @antmicro has recently been focused on making the Surelog+UHDM -> Yosys -> OpenROAD flow more seamlessly usable -- might want to check out my diagram at https://j.mp/sv-flow-diagram too.

ChristianB12, 2021-07-21:
> You might want to consider [Surelog](https://github.com/chipsalliance/Surelog) + [UHDM](https://github.com/chipsalliance/uhdm) as an option for [parsing SystemVerilog for usage with [...]

Thanks. Looks like this should work

zachjs, 2021-07-21:
Based on my (limited) understanding of OpenLane, it sounds like the end goal here is synthesis, but I (perhaps naively) don't believe the above code is synthesizable. Regardless of what you decide to do, I'm eager to better understand the use case.

ChristianB12, 2021-07-22:
> Based on my (limited) understanding of OpenLane, it sounds like the end goal here is synthesis, but I (perhaps naively) don't believe the above code is synthesizable. Regardless of what you decide to do, I'm eager to better understand the use case.

I'm essentially following the diagram @mithro shared, but instead using one of the designs on the left, I'm trying to use a SoC generated by [Chipyard](https://chipyard.readthedocs.io/en/latest/).

And I believe you're right about it not being synthesizable. I'm still trying to figure out what to do about...

zachjs, 2021-07-22:
Given that the module is not synthesizable, my inexperienced recommendation would be to remove/exclude it from your flow. Presumably the source will reference some "real" DRAM module, rather than `SimDRAM`.

zachjs, 2021-07-27:
Are you still facing an issue here? sv2v should be a feasible option for such synthesis flows.

zachjs, 2021-08-13:
Given the inactivity, I am closing this issue. Please feel free to reopen or file a new one if appropriate!

closed 2021-08-13 by zachjs