zachjs/sv2v #159: sv2v hangs on CVA6

flaviens, 2021-07-22:
Hi Zach,

I'm really sorry for opening a concurrent issue :disappointed: But at the same time, that's the best way to improve sv2v :slightly_smiling_face:

When applying sv2v on a Pickled version of CVA6, sv2v hangs (or maybe takes an usual very, very long time).

Steps to reproduce:
```
git clone https://github.com/openhwgroup/cva6.git
cd cva6
git submodule update --init --recursive
bender sources -f -t synthesis > bender.sources
sed -i "s/synopsys_sram/util\/sram/g" bender.sources
morty -f bender.sources -DVERILATOR -DSYNTHESIS > pickled.sv
sv2v pickled.sv
```

You can also download the file directly from here if you prefer: https://github.com/flaviens/pickled_cva6

zachjs, 2021-07-22:
Thanks for sending along this report! There was indeed an issue which could cause certain expression simplification rules to shuffle some unsimplified numbers endlessly. I've pushed a workaround for this issue, but I intend to beef up sv2v's ability to constant fold in the future, which should obviate the workaround.

flaviens, 2021-07-23:
Great, that indeed solved the problem!

Congrats again for your reactivity! :+1:

closed 2021-07-23 by flaviens