zachjs/sv2v #160: Support for datatypes on the wire nettypes

chiraag, 2021-07-23:
I was trying to work on a design with the following syntax:
```sv
wire logic x = 1'b0;
```

This is error message I get:
```bash
sv2v: takeIdent didn't find identifier: [DTType rtl/top.sv:11:12 logic,DTIdent rtl/top.sv:11:14 "x",DTAsgn rtl/top.sv:11:20 = Nothing 1'b0]
CallStack (from HasCallStack):
  error, called at src/Language/SystemVerilog/Parser/ParseDecl.hs:496:20 in main:Language.SystemVerilog.Parser.ParseDecl
```

I believe this is valid syntax.

zachjs, 2021-07-23:
What version are you using? This feature is implemented on master.

chiraag, 2021-07-23:
I was using the v0.0.7 binary release? Have there been a ton of updates since then?
If yes, I can try using the binary from the latest Github build action.

zachjs, 2021-07-23:
Yes, please give that a shot. v0.0.7 is pretty old at this point. I will look into making at least quarterly releases in the future.

chiraag, 2021-07-23:
Works with 1e6fa7b

closed 2021-07-23 by chiraag