zachjs/sv2v #161: Error inlining an instance with an interface

chiraag, 2021-07-23:
I am trying to work through the following error message that I got running `sv2v`.

```
sv2v: inlining instance "test_u" of module "test" would make expression "ifc.rdy" used in "test_u" resolvable when it wasn't previously
CallStack (from HasCallStack):
  error, called at src/Convert/Interface.hs:502:21 in main:Convert.Interface
```

Could you help me understand the underlying issue? Let me know if there an IRC channel or some other place better suited for questions like this.

zachjs, 2021-07-23:
There's a good chance the error here is incorrect. Based on the error message, I found some cases where it could be raised when it shouldn't. I created a branch called `interface-check-extra` which has a probable fix. I have some revisions I'd like to make later tonight before pushing it to master, but it may be worth giving a shot in the meantime.

zachjs, 2021-07-24:
I've pushed a slightly updated version of this fix to master and removed the temporary branch. Please let me know what you find!

chiraag, 2021-07-24:
The conversion worked for me with 1311e44. I have not simulated the converted design yet. I will close this issue and reopen another if there is a simulation mismatch. Thanks a lot for helping out with this!

closed 2021-07-24 by chiraag