zachjs/sv2v #162: Streaming operator issue in combination with Verilator

flaviens, 2021-07-26:
Hello Zach,

It's me again for a new issue! :slightly_smiling_face:

There seems to be an issue with the streaming operator when chaining sv2v with Verilator.
Although simple tests with the streaming operator pass, the following does not:

```
git clone git@github.com:flaviens/streaming_operator_issue.git
cd streaming_operator_issue
make
```

Precisely, sv2v delivers an output rejected by Verilator:
```verilog
%Error-UNSUPPORTED: riscv_alu_div.v:71:25: Unsupported: left < right of bit extract: -1<0
                                         : ... In instance riscv_alu_div
   71 |     _sv2v_strm_55E18_out[0+:C_WIDTH % 1] = _sv2v_strm_55E18_inp[_sv2v_strm_55E18_idx+:C_WIDTH % 1];
      |                         ^
                    ... For error description see https://verilator.org/warn/UNSUPPORTED?v=4.210
%Error: Internal Error: riscv_alu_div.v:71:64: ../V3Width.cpp:5025: Under node SEL has no expected width?? Missing Visitor func?
                                             : ... In instance riscv_alu_div
   71 |     _sv2v_strm_55E18_out[0+:C_WIDTH % 1] = _sv2v_strm_55E18_inp[_sv2v_strm_55E18_idx+:C_WIDTH % 1];
      |                                                                ^
```

The problem seems to correspond to line 64 of the source file.

zachjs, 2021-07-26:
Thanks for pointing this out! That line was guarded with a condition like `if ((W % 1) > 0)`. While `iverilog` seems to elaborate this away, `verilator` and others do not. I've added simple logic to ensure sv2v should constant fold and remove these statements.

flaviens, 2021-07-27:
Nice, that seems to work! :+1:

closed 2021-07-27 by flaviens