zachjs/sv2v #164: Package function + pickle + Verilator

flaviens, 2021-07-27:
Hi Zach,

It's me again :stuck_out_tongue_closed_eyes: 

With a pickle PULPissimo + Verilator again, I encounter an issue regarding package functions.
Now sv2v successfully treats PULPissimo, but apparently Verilator notices that a function is out of scope.

```
%Error: soc_domain.v:28606:19: Can't find definition of task/function: 'cf_math_pkg_idx_width'
28606 |   .default_idx_i({cf_math_pkg_idx_width(NoApbSlaves) {1'sb0}})
      |                   ^~~~~~~~~~~~~~~~~~~~~
%Error: soc_domain.v:110316:24: Can't find definition of task/function: 'cf_math_pkg_idx_width'
110316 |        .default_idx_i({cf_math_pkg_idx_width(NR_OUTPUTS) {1'sb0}})
       |                        ^~~~~~~~~~~~~~~~~~~~~
%Error: soc_domain.v:110494:24: Can't find definition of task/function: 'cf_math_pkg_idx_width'
110494 |        .default_idx_i({cf_math_pkg_idx_width(NR_OUTPUTS) {1'sb0}})
       |                        ^~~~~~~~~~~~~~~~~~~~~
```

You can reproduce the error by cloning [this repo](https://github.com/flaviens/pkg_function_issue).
As usual, run `make` to reproduce the issue, and `make reference` to successfully compile directly the sv source.

zachjs, 2021-07-28:
This is related to a limitation with the way the unbased unsized conversion generates the width for these expressions in ports. I have a sketch for a better strategy that does not have this limitation, but it's not quite ready yet. I anticipate having it ready tomorrow or the next day.

I've already spotted and fixed two other issues in this file. The first would caused genvars shadowing other genvars of the same name to be incorrectly removed. The second prevented implicitly-typed output ports from becoming regs.

zachjs, 2021-07-29:
Please give the `uu-inline` branch a shot!

flaviens, 2021-07-29:
Great job! The `uu-inline` version indeed works now with Verilator!

I just created [a new issue](https://github.com/YosysHQ/yosys/issues/2898) in the Yosys repo because Yosys cannot parse the output.
Maybe you want to have a look!

zachjs, 2021-07-30:
An updated version of the the `uu-inline` changes has landed in master, and the branch has been removed.

flaviens, 2021-07-30:
Perfect

closed 2021-07-30 by flaviens