zachjs/sv2v #168: One issue was good in V0.07 but failed in V0.08

stitchlibar, 2021-08-17:
It complains that 
sv2v: test.sv:14:51: Parse error: declaration missing type information
CallStack (from HasCallStack):
  error, called at src/Language/SystemVerilog/Parser/ParseDecl.hs:599:26 in main:Language.SystemVerilog.Parser.ParseDecl

function automatic signed [2:0] p4vs(input en,err,[1:0]d);
reg signed [2:0] sym;
   begin
      case(d[1:0])
	2'b00: sym = 3'b101;
	default: sym = 3'b111;
      endcase // case (d[1:0])
      if(!en) p4vs  = 3'd0;
      else p4vs = err? sym : -sym;
   end
endfunction

zachjs, 2021-08-17:
Thanks for this report! This should be fixed in 7ccab1c70a1236a078e9f913589de8a5a22d391b. Please let me know if it works for you!

stitchlibar, 2021-08-19:
Fixed in the latest version.

closed 2021-08-19 by stitchlibar