zachjs/sv2v #169: Convert has error in Scoper.hs

stitchlibar, 2021-08-17:
V0.0.7 has error in Convert showing
src/Convert/Scoper.hs:453:17-59: Non-exhaustive patterns in Variable Local
         newType
         newName
         [] 
         Nil

zachjs, 2021-08-17:
Is it possible this is related to https://github.com/zachjs/sv2v/issues/157#issuecomment-883203231? Have you tried with v0.0.8 and/or the latest?

stitchlibar, 2021-08-17:
V0.0.8 can't run to this point. It already errors out with issue #168

zachjs, 2021-08-17:
Are you able to try with a version built from the latest master?

stitchlibar, 2021-08-17:
I will try. I can get the latest by Code->Download ZIP, right?

zachjs, 2021-08-17:
You can do it that way, but my recommendation would be to `git clone` the repo, following the instructions in the readme. Then you can `git pull` as new changes are pushed, and incremental builds should be a bit faster.

stitchlibar, 2021-08-17:
Thanks. I will try the git way in the future. Seems the latest release fixed the issue. Thank you!

zachjs, 2021-08-17:
That's great to hear! For any issues you see fixed, I'd appreciate it if you could update/close them as appropriate, provided the fix is working on your end.

stitchlibar, 2021-08-19:
Verified

closed 2021-08-19 by stitchlibar