zachjs/sv2v #171: 2nd expression of procedural for-loop is not constant from Yosys

stitchlibar, 2021-08-17:
Seems Yosys doesn't like the result of SV2V. It complains 
2nd expression of procedural for-loop is not constant from Yosys

	 for(int i=7;i>=sign;i--) begin
	    somefun[i] 	= 1'b0;
	 end

Check the attached SysV line 18 for detail.
[test_copy.txt](https://github.com/zachjs/sv2v/files/7003445/test_copy.txt)

stitchlibar, 2021-08-18:
On correction. The complained non-constant is actually a constant when the parent is passing the argument. Please discard test_copy.txt and check test_copy2.txt
[test_copy2.txt](https://github.com/zachjs/sv2v/files/7003619/test_copy2.txt)

zachjs, 2021-08-18:
This is a limitation of Yosys. You may be able to work around it by replacing
```SystemVerilog
for(int i=7;i>=sign;i--) begin
    somefun[i] = 1'b0;
end
```
with
```Verilog
somefun[sign:7] = 1'b0;
```

The latest Yosys master has some improvements to the handling of constant function arguments, but they don't help in this particular case. Additionally, if you use the latest Yosys master, I believe you will run into the issue fixed by https://github.com/YosysHQ/yosys/pull/2716.

stitchlibar, 2021-08-18:
Great suggest! Is it ok to be somefun[7:sign] = '0;

zachjs, 2021-08-18:
Correction: You should probably use `somefun[sign-:7] = '0;` as the LRM does not permit non-constant expressions for non-indexed range bounds. You could also consider using `localparam`s to define `SIGN` if the value is truly constant.

stitchlibar, 2021-08-19:
It works.

closed 2021-08-19 by stitchlibar