zachjs/sv2v #172: Error in two dimension array conversion

stitchlibar, 2021-08-18:
SV2V leaves some two dimension arrays not converted. And it causes assignment error when a two-dimension array is assigned to one-dimension bus.
// Only one two-d array is converted, and leave the other 4 untouched
[test_copy3.txt](https://github.com/zachjs/sv2v/files/7004182/test_copy3.txt)

       wire [8:0] isiga;
        wire [2:0] isigb [2:0];
        wire [2:0] isigc [2:0];
        wire [2:0] isigd [2:0];
        wire [2:0] isige [2:0];
 
        assign twod = (in0[0] ? isiga : (in0[1] ? isigb : (in1[0] ? isigc : (in1[1] ? isigd : isige))));

zachjs, 2021-08-18:
Thanks for reporting this issue! It should be fixed as of 9acdb848c9027c2acd95289618669c6a352f3c29.

stitchlibar, 2021-08-19:
Works great! Thanks!

closed 2021-08-19 by stitchlibar