zachjs/sv2v #173: Sensitive list "@ *" is not recognizable

stitchlibar, 2021-08-19:
The app fails on @ *,  Parse error: unexpected token '*' (Sym_aster)

always @ *
 begin
   a = b & c;
end

Other commercial tool can parse it.

zachjs, 2021-08-19:
Thank you for pointing this out! 77ee49a80e9ea64988dae3750c20f5912d2a17ab should fix it.

stitchlibar, 2021-08-19:
Thanks. But when I build the latest release, it takes much longer time than before.
And ghc used up all CPU resources.

zachjs, 2021-08-20:
Unfortunately, the parser is the slowest to compile out of sv2v's modules. My only suggestion for the moment is to let it run. I wouldn't be surprised if this could take a couple minutes on certain machines. However, you seem to have been able to compile that file in the past, and the recent changes should have made the situation any worse. I will still look into this further, because the slowdown appears to be much worse with GHC 9.X, which I would eventually like to upgrade to.

zachjs, 2021-08-20:
Please let me know if you were able to get that build to complete! I did just push a small change which should make recompilation more efficient (fewer threads) without making it slower. I've also started looking into the GHC 9.X issue, and it appears unrelated.