zachjs/sv2v #174: Unsigned to signed cast issue

stitchlibar, 2021-08-20:
I think the translation of unsigned to signed has some issue
SV
assign p4syn = 4'(p4vs(en, err, din));
to V, the input is actually 3 bits not four bits

	function automatic signed [3:0] sv2v_cast_4_signed;
		**input reg signed [3:0] inp;**
		sv2v_cast_4_signed = inp;
	endfunction
	assign p4syn = sv2v_cast_4_signed(myparam_p4vs(en, err, din));

Yosys errors out with this, not sure it's directly related
ERROR: Assert `arg->is_signed == sig.as_wire()->is_signed' failed in frontends/ast/genrtlil.cc:1858.

[test_copy4.txt](https://github.com/zachjs/sv2v/files/7024228/test_copy4.txt)

stitchlibar, 2021-08-20:
The yosys issue is caused by signed signal connections. The bus delimiter [3:0] should be removed to pass yosys. 
	   .p4syn			(p4syn[3:0]),
	   .p4syn2			(p4syn2[3:0]),

stitchlibar, 2021-08-20:
Sorry, I would like to take back. Seems the netlist is correct in the signed bit extension. So it's not an issue.

closed 2021-08-20 by stitchlibar

zachjs, 2021-08-21:
That Yosys assertion failure is tracked in https://github.com/YosysHQ/yosys/issues/2654, and has a pending fix: https://github.com/YosysHQ/yosys/pull/2716.