zachjs/sv2v #175: wire assignment is removed

stitchlibar, 2021-08-22:
The following module is empty after conversion. SV2V ignores "wire estart = start&enable;". Changing to assign would work.
[test_wireonly.txt](https://github.com/zachjs/sv2v/files/7027711/test_wireonly.txt)

module wiremod #(parameter abc = 1) (/*AUTOARG*/
// Outputs
estart,
// Inputs
enable, start
);
input enable, start;
output estart;
//assign works, wire doesn't
//assign   estart  = start & enable;
wire   estart  = start & enable;
endmodule

zachjs, 2021-08-23:
sv2v does not yet have good handling for port declarations split in this wag. It may take a few days before I have this properly supported.

stitchlibar, 2021-08-23:
It is loosely written, if Design Compiler never spoiled the designers by throwing errors in the beginning.
Thanks and take your time.

zachjs, 2021-08-29:
Yes, those two formulations are equivalent.

zachjs, 2021-09-07:
Sorry for the delay in getting this issue fixed! It should be fixed in 95c2bc996c2472a749fb23b2521b88ef6e041584. Please let me know if it works for you!

stitchlibar, 2021-09-08:
Yes, it has been fixed. Great work!

closed 2021-09-08 by stitchlibar