zachjs/sv2v #176: ERROR: hGetContents: invalid arguments (invalid byte sequence)

stitchlibar, 2021-08-25:
What is this error? Only happens in Linux compile sv2v, cygwin sv2v is good. The SV file format is utf-8-unix.

zachjs, 2021-08-25:
`hGetContents` is a Haskell library function. In my experience, it generally assumes UTF-8, but it seems that may not always be the case. I think I could add a character encoding option to sv2v, or try to make it more tolerant in general. Are you able to isolate the file or byte sequence in question?

https://hackage.haskell.org/package/base-4.15.0.0/docs/System-IO.html

stitchlibar, 2021-08-25:
Any non ascii in the source file would cause the error. For example "‘" in the following comment.

//  ‘ test parameter something = other? 1 : 0
package myparam;
localparam [7:0] TTSEE = 8'h21;
localparam [7:0] TTSFF = 8'h21;
localparam [7:0] TTSGG = 8'h21;
localparam WIDTH  = 8;
typedef logic [WIDTH-1:0] aat[2:0];

zachjs, 2021-08-30:
I pushed d335d2ff25a5ffd4f49dce5cbedcae00e3115b51 to master, which should tolerate transcoding failures. Please let me know if that works for you!

stitchlibar, 2021-09-02:
Yes, the issue has been fixed by the new release.

closed 2021-09-02 by stitchlibar