zachjs/sv2v #179: Add support for assertion

saw235, 2021-09-11:
lowRISC ibex uses sv2v for compatibility with opensource tool, but also includes their own workaround for assertions via macros, which is able to work with both yosys and symbiyosys. 

See both
[assert_yosys_macros](https://github.com/saw235/opentitan/blob/master/hw/ip/prim/rtl/prim_assert_yosys_macros.svh)
[ibex_assertions](https://github.com/lowRISC/ibex/blob/master/formal/icache/formal_tb.sv`)

We can extend it to generate the macro item and assertions with the method above as examples. The only caveat is that it is not clear how deferred assertion "assert #0" can be achieved.

zachjs, 2021-09-11:
It sounds like you'd like sv2v to preserve assertions in the source so that these other tools can read and check them. Is this accurate?

There is currently an option to disable the automatic removal of assertions via `-E assert`/`--exclude assert`. As you mention, it is possible that some additional work may be necessary, so please let me know how far this gets you!

zachjs, 2021-10-03:
@saw235 Any update on this issue? I'm eager to better understand and support your use case!

zachjs, 2021-11-10:
Given the inactivity, I am closing this issue. Please feel free to update it if necessary!

closed 2021-11-10 by zachjs