zachjs/sv2v #180: comment after 'end' without space causes error

stitchlibar, 2021-09-13:
If a comment is right after 'end' keyword without space, it will fail sv2v
always @(posedge clk or negedge rst) begin
   if(!rst) begin
      comp_in_dlyi <= 5'b0;
      confdly 	   <= 1'b0;
   end
   else if(cmpen) begin 
      comp_in_dlyi <= 5'b10001;
      confdly 	   <= 1'b1;
   end else begin
      comp_in_dlyi <= comp_in;
      comfdly 	   <= conf[1];
   end//end of main
end

Full test file is here
[san1234.txt](https://github.com/zachjs/sv2v/files/7155666/san1234.txt)

zachjs, 2021-09-14:
Thank you for finding and reporting this issue! It should be fixed as of e169c907f4634fe0e28e7caa6f2e80385cd5f4ca.

stitchlibar, 2021-09-15:
It has been fixed.

closed 2021-09-15 by stitchlibar