zachjs/sv2v #181: Unpacked dimensions on types from parameterized interface

zegervdv, 2021-09-13:
related to #97 and #96 

I'm using an interface to define a parameterized struct. Then I instantiate an unpacked array of that type which returns an error:
```
sv2v: non-vector type foo_port.test cannot have packed dimesions:[(0,2 - 1)]
CallStack (from HasCallStack):
  error, called at src/Language/SystemVerilog/AST/Type.hs:114:14 in main:Language.SystemVerilog.AST.Type
```
Without the unpacked dimensions this works fine.
Seems like the array is mistakenly parsed as packed dimensions?

Minimal test example:
``` systemverilog
interface foo #(
  parameter int unsigned width
);

  logic [width-1:0] test;

  modport m(input test);
  modport s(output test);
endinterface

module bar (
  foo.m foo_port
);
  typedef foo_port.test foo_test;

  foo_test signal [2];
endmodule

module top (
  input logic[10:0] flat
);

foo#(.width(11)) itf();
bar mod (.foo_port(itf.m));
endmodule
```

zachjs, 2021-09-13:
While there may be an issue here, I'm not convinced this test case is valid. `foo_port.test` is a variable, not a typename. Is this intentional?

zegervdv, 2021-09-14:
Oops, I messed up the testcase :)

This should be a valid scenario.
It still shows the same error though.
``` systemverilog
interface foo #(
  parameter int unsigned width
);

  typedef struct packed {
    logic [width-1:0] test;
  } test_struct;

  test_struct test;

  modport m(input test);
  modport s(output test);
endinterface

module bar (
  foo.m foo_port
);
  typedef foo_port.test_struct foo_test;

  foo_test signal [2];
endmodule

module top (
  input logic[10:0] flat
);

foo#(.width(11)) itf();
bar mod (.foo_port(itf.m));
endmodule
```

zachjs, 2021-09-19:
Thanks for filing this issue! I believe it should now be fixed. Please let me know if it works for you!

zegervdv, 2021-09-21:
Thanks for the fix! I can confirm it works for my usecase too.

closed 2021-09-21 by zegervdv