zachjs/sv2v #186: 0.0.9 hackage release

newAM, 2021-10-02:
Thank you for creating `sv2v`, it has been very useful for me!

I was looking at updating `sv2v` to 0.0.9 in NixOS.  NixOS uses hackage releases, and 0.0.9 is still a candidate release over there: https://hackage.haskell.org/package/sv2v-0.0.9/candidate

This is my first time looking at haskell package management, forgive me if I am reading this wrong, but it looks like version 0.0.9 needs to be published for it to be available as a non-candidate release on hackage.  Are you able to do this so I can update `sv2v` in NixOS?

zachjs, 2021-10-02:
Thank you for the reminder! Indeed, I forgot to publish that candidate, but it should now be fixed. On a separate note, please let me know if there is anything I can do to make package maintenance easier!

newAM, 2021-10-03:
I don't think there is anything else to make this easier :)

Thank you for publishing the release!

closed 2021-10-03 by newAM