zachjs/sv2v #188: Decimal literals can't always be constant-folded

npatil2, 2021-11-01:
The code below illegally optimizes out the 0 in (a + b + 0).

I think the problem is that literal 0 needs to be interpreted as 32'sd0, and not participate in constant folding (but then I am not sure what the purpose of constant folding would even be). The manual leaves the implied width of unsized integer literals to the tool, but requires that it be at least 32 bits.

```
$ cat ab0.sv 
module ab0;
    logic [15:0] a, b;
    logic carryout;
    assign carryout = (a + b + 0) >> 16;
endmodule

$ sv2v ab0.sv 
module ab0;
        wire [15:0] a;
        wire [15:0] b;
        wire carryout;
        assign carryout = (a + b) >> 16;
endmodule
```

Excerpts from relevant sections in Std 1800-2017:

**5.7.1  Integer literal constants**

The number of bits that make up an unsized number (which is a simple decimal number or a number with a base specifier but no size specification) shall be at least 32. Unsized unsigned literal constants where the high-order bit is unknown (X or x) or three-state (Z or z) shall be extended to the size of the expression containing the literal constant. 

**11.6.2 Example of expression bit-length problem**

During the evaluation of an expression, interim results shall take the size of the largest operand (in case of an assignment, this also includes the left-hand side). Care has to be taken to prevent loss of a significant bit during expression evaluation. The following example describes how the bit lengths of the operands could result in the loss of a significant bit.

Given the following declarations:

`logic [15:0] a, b, answer; // 16-bit variables`

the intent is to evaluate the expression

`answer = (a + b) >> 1; // will not work properly`

where a and b are to be added, which can result in an overflow, and then shifted right by 1 bit to preserve the carry bit in the 16-bit answer. 

A problem arises, however, because all operands in the expression are of a 16-bit width. Therefore, the expression (a + b) produces an interim result that is only 16 bits wide, thus losing the carry bit before the evaluation performs the 1-bit right shift operation.

The solution is to force the expression (a + b) to evaluate using at least 17 bits. For example, adding an integer value of 0 to the expression will cause the evaluation to be performed using the bit size of integers. The following example will produce the intended result:

`answer = (a + b + 0) >> 1; // will work correctly`

zachjs, 2021-11-15:
Thank you for filing this issue! I completely agree that the simplification sv2v is performing is illegally changing the behavior of the design. While it is possible to make the width extension explicit using something like `17'(a + b) >> 1`, I'd like to fix sv2v's behavior here.

I've created a branch called `clip` which avoids removal of `+ 0` and `* 1`, while still simplifying the intended conversion artifacts. Please try it out and let me know if it works for you!

zachjs, 2021-12-10:
I've merged the fix I described above onto the main branch. Given the lack of activity, I am closing this issue. Please reopen it or file a new one if necessary!

closed 2021-12-10 by zachjs