zachjs/sv2v #189: Processing macros alone

rahulraveendran15-coder, 2021-11-08:
Hi,

I would like to know is it possible to use the passed define macros in a file and not elaborate the `include file in the converted file?

Ex:
```
`include "sample.svh"
module test();
`ifdef SAMPLE
parameter test = 2;
`else
parameter test = 3;
`endif
```

This SAMPLE  will be defined in the sample.svh along with some other parameter values or defines

I want  output like this with SAMPLE  considered as defines in the file:

```
`include "sample.svh"
module test();
parameter test = 2;

```

zachjs, 2021-11-08:
In this scenario, are you expecting sv2v to still convert the rest of the contents of the file? What if the contents of the include are relevant to the conversion (i.e., typedefs)? In any event, I'd greatly appreciate it if you could help me understand your overall use case. Why do you need includes to be preserved? How does this fit in to the rest of your flow?

rahulraveendran15-coder, 2021-11-08:
I have  some files which has so many ifdefs. So it makes my RTL files too bulky. At the same time my defines are defined inside the include files whihc I mentioned bebove.
The reason why I want to preserve these include files as it is because this also contains some define values like `define ABC 4'h4. So if I want to change the ABC  value later  I need to modify several files which has this `ABC  value used.

zachjs, 2021-11-15:
Unfortunately, it's probably infeasible to process some macros and not others during preprocessing, as macros are inherently ambiguous in parsing (and conversion!) until they are expanded. However, there may be a reasonable workaround.

For definitions which are simply expressions, it should be possible to use `localparam` rather than `` `define ``. For example, with the following files, you could only pass `b.sv` and `c.sv` through sv2v (notice that they don't include `a.sv`), and then pass `a.sv` and sv2v's output to downstream tools, allowing you to tweak the isolated parameters as needed.
```SystemVerilog
// File: a.sv
localparam ABC = 4'h4;
localparam XYZ = ABC * 2;
...
```
```SystemVerilog
// File: b.sv
module mod;
...
assign x = y & ABC;
...
endmodule
```
```SystemVerilog
// File: c.sv
module thing;
...
logic [ABC - 1:0] q;
...
endmodule
```

If you have macros in your header which can't be isolated in this way, could you provide a concrete example? This should help me further understand your use case.

zachjs, 2021-12-10:
@rahulraveendran15-coder We you able to get something working on your end? I'm eager to help if I can!

closed 2022-01-10 by zachjs