zachjs/sv2v #191: Unable to convert an array of named constants in an enum

kauser-rl, 2021-12-08:
Try to convert a legal SV `typedef enum` below.

```
package test1_pkg;
   typedef enum logic[8:0] {
     A[2]   = 9'h004,
     B[2]   = 9'h006,
     C[16]  = 9'h0a0,
     D      = 9'h123,
     E[4]   = 9'h124,
     UNUSED = 9'hF00
   } test_t;

endpackage // test1_pkg
```

zachjs, 2021-12-10:
sv2v doesn't yet support this feature, but I can add it! I'll update here within a few days.

zachjs, 2021-12-10:
I've pushed support for enumerated type ranges. Please give it a shot and let me know if it works for you!

Note that `9'hF00` in the above example is rejected by sv2v because the digits make up 12 bits, in excess of the explicitly given width of 9.

kauser-rl, 2021-12-13:
I seem to have hit an error trying to build from the latest cloned version of the repo. Is it possible to generate a binary for `Ubuntu 20.04`?

closed 2022-01-10 by zachjs