zachjs/sv2v #194: Question: how do you handle relating error messages to the original systemverilog source code?

hughperkins, 2022-02-25:
Question: how do you handle relating error messages to the original systemverilog source code?

Like, an issue I find with using generators (of which, this project is generating verilog, given systemverilog), is that one gets error messages in the generated language, not in the language one is actually writing code in. This basically forces one to read the generated code, and have to work with the generated code.

An example of a possible solution could be a tool which takes an error in the generated verilog, and converts it into an appropriate message for the original systemverilog code. For example, during verilog generation, the original line numbers and so on could be stored somehow/somewhere, and then could be read by such a tool.

Just wondering what approach is  being used for sv2v?

zachjs, 2022-02-25:
If there are errors in the SystemVerilog source, my intention is for either sv2v to produce a useful error, or for the error to be visible in a reasonable downstream tool like Yosys. Although there is not currently an "automated" way to "convert" the source positions in downstream errors, sv2v does leave source position traces and some other information in the generated output when run in verbose mode. These can be used to relate errors to the original SystemVerilog source.

zachjs, 2022-03-14:
@hughperkins Has your question been fully addressed?

hughperkins, 2022-03-14:
You've technically answered the question. I kind of would like to create a request to automatically relate error messages back to the original sv. Although, I suppose you could argue that:
- any errors in syntax etc would be caught by sv2v anyway?
- typically, verilog programs dont really "crash" as such, with a stack trace, like once theyve compiled, they just dont do what we want, and to debug that, things like $display work well, and those will work unchanged in sv2v anyway?

hughperkins, 2022-05-09:
So, in practice, circling around, it is easily possible to inadvertently construct programs which pass through `sv2v` compilation with no errors, but the resulting verilog does have errors, and then I have to open the verilog file, and figure out what is going on, e.g. https://github.com/zachjs/sv2v/issues/206

georgerennie, 2022-05-20:
Yosys has the ability to read `src` attributes, e.g.
```verilog
(* src = "test.v:24.1-30.10" *)
module top(a, b);
  (* src = "test.v:25.8-25.9" *)
  input a;
  wire a;
  (* src = "test.v:26.9-26.10" *)
  output b;
  wire b;
  assign b = a;
endmodule
```

Seeing as the trace statements are already generating something similar to this, how hard would it be to add ability for sv2v to generate these attributes? Alternatively there is \`line in verilog which should potentially have better support by other tools (the src attribute is yosys specific afaik) but is more intended for simply copying and pasting lines, not the shifting of content that can occur in sv2v due to the ast representation. It would be quite nice if source line numbers could be preserved for things like symbiyosys that references failing assertions by their `src` attribute