zachjs/sv2v #196: Unary increment/decrement in assignment support

mikhailmoiseev, 2022-03-09:
sv2v reports ""Parse error: unexpected token '++' (Sym_plus_plus)" for unary increment in expression:
```verilog
a = b++;
a = ++b;
a = (b++) + c;
...
```
The same for decrement.

Are you planning to support that?

zachjs, 2022-03-09:
Yes, I am interested in adding support. You can expect an update within a few days.

zachjs, 2022-03-10:
> sv2v is really great tool. I have checked it with 300+ complex SV files from [Intel Compiler for SystemC (ICSC)](https://github.com/intel/systemc-compiler) testbench, no major issues found.

That's great to hear!

> Specially, I am impressed with multi-return functions to task conversion.

Out of curiosity, are you referring to the conversion of conversion of `return`, `continue`, and `break`, or something else?

> Do you mind if I add a link to this repo from ICSC GitHub wiki?

Not at all. Thank you!

mikhailmoiseev, 2022-03-11:
> Out of curiosity, are you referring to the conversion of conversion of return, continue, and break, or something else?

Yes, it is about conversion of multiple returns from function. ICSC translates multi-state FSM in SystemC threads into awlays_comb with function call, where the function contains case of FSM state variable and multiple returns from it. So, that is important.

zachjs, 2022-03-13:
I just pushed support for assignments within expressions. Please let me know if it works well for you!

mikhailmoiseev, 2022-03-15:
That works well. Thank you!

zachjs, 2022-03-16:
I'm glad it's working for you!

closed 2022-03-16 by zachjs