zachjs/sv2v #200: Feature request: Preserve macros in the design.

kauser-rl, 2022-03-16:
Hi,
I believe this isn't currently possible. I would like to convert SV to Verilog, but preserve all of the macros in the design (while converting the portion within the macros to Verilog too). I have provided an example below. Maybe this is already possible?

Original SV code.
```
module b (
    input wire logic          inp1_i,
    input wire logic          inp2_i,
`ifdef OPTIONAL
    output     logic [1:0]    out_opt_o,
`endif
    output     logic          out1_o,
    output     logic          out2_o
   );
endmodule // b
```

If I simply convert this to Verilog with the `OPTIONAL` macro set.
```
module b (
	inp1_i,
	inp2_i,
	out_opt_o,
	out1_o,
	out2_o
);
	input wire inp1_i;
	input wire inp2_i;
	output wire [1:0] out_opt_o;
	output wire out1_o;
	output wire out2_o;
endmodule
```
But what I would like is.
```
module b (
	inp1_i,
	inp2_i,
`ifdef OPTIONAL
	out_opt_o,
`endif
	out1_o,
	out2_o
);
	input wire inp1_i;
	input wire inp2_i;
`ifdef OPTIONAL
	output wire [1:0] out_opt_o;
`endif
	output wire out1_o;
	output wire out2_o;
endmodule
```

I appreciate this is a bespoke request and may not be trivial to add.

zachjs, 2022-03-16:
I think the discussion in #76 is relevant. Can you check that out?

kauser-rl, 2022-03-16:
I will give the suggestion a try and get back to you. Thanks!

zachjs, 2022-04-09:
@kauser-rl Do you have any update here?

kauser-rl, 2022-04-11:
The suggestion to use `diff` works, but that makes me nervous. I have to manually review the output every time I re-spin just because I'm paranoid. But that is life.

closed 2022-04-11 by kauser-rl