zachjs/sv2v #201: interface instance syntax error

MoonbaseOtago, 2022-03-16:
interface instances without connection lists fail with a syntax error - this is valid
(for examples: https://www.chipverify.com/systemverilog/systemverilog-interface )

For example if I  create:

    interface fred;
        bit molly;
    endinterface

    module .....

           fred inst;
     ...
     endmodule

I get an error, if I instantiate "fred inst{}; I don't get the syntax error

zachjs, 2022-03-16:
As far as I can tell, the connections on an instantiation are not optional. From the IEEE 1800-2017 grammar:
```
interface_instantiation ::=  interface_identifier [ parameter_value_assignment ] hierarchical_instance { , hierarchical_instance } ;
hierarchical_instance ::= name_of_instance ( [ list_of_port_connections ] )
name_of_instance ::= instance_identifier { unpacked_dimension }
```

I tried the following in all of the commercial tools available on edaplayground.com, and none of them allow it. A few produce an error that suggests parens could be missing.
```SystemVerilog
interface intf;
endinterface

module mod(intf i);
endmodule

module top;
  intf i;
  mod m(i);
endmodule
```

sv2v produces an error like: `Modport not in port list: intf i. Is this an interface missing a port list?` Are there tools which allow this syntax?

zachjs, 2022-03-18:
> https://www.chipverify.com/systemverilog/systemverilog-interface

The relevant example I found there is also not accepted by any commercial tool as far as I can tell. There is also at least one _other_ typo, so I'm not convinced that these examples should be taken as gospel.

closed 2022-05-31 by zachjs