zachjs/sv2v #204: Extra semicolon emitted

jiegec, 2022-03-29:
SystemVerilog:

```sv
module test (
  output [3:0] test
);
  int num;
  if (num == 1) begin
    assign test = 2;
  end else if (num == 2) begin
  end else begin
    assign test = 3;
  end

endmodule
```

Translated:

```verilog
module test (test);
	output wire [3:0] test;
	wire signed [31:0] num;
	generate
		if (num == 1) begin : genblk1
			assign test = 2;
		end
		else if (num == 2) begin
			;
		end
		else begin : genblk1
			assign test = 3;
		end
	endgenerate
endmodule

```

Yosys output:

```
1. Executing Verilog-2005 frontend: test.v
Parsing SystemVerilog input from `test.v' to AST representation.
test.v:9: ERROR: syntax error, unexpected ';'
```

zachjs, 2022-03-29:
The particular error you are hitting in Yosys is due to the use of an old version. I believe this was fixed in https://github.com/YosysHQ/yosys/pull/2028. When using a more recent version, you'll instead get `issue.v:5: ERROR: Condition for generate if is not constant!`, which is accurate because `num` is not a constant in the provided example.

jiegec, 2022-03-29:
Thanks, it works with yosys master.

jiegec, 2022-03-29:
Thanks for you great project, I am able to convert fpnew to verilog and then synthesize with yosys! :smile: 🎉