zachjs/sv2v #205: Generated verilog code cannot be read by yosys

mmxsrup, 2022-04-27:
I get an error when I convert the module `GetInsnPtr` shown below to verilog with sv2v and read it in yosys. The version of sv2v is `v0.0.9-27-g5e17ef0`. Is the sv2v conversion correct? Or is it a bug in yosys?  I have made a similar issue on yosys (https://github.com/YosysHQ/yosys/issues/3291).

Original code (https://github.com/rsd-devel/rsd/blob/8777d59e0b9aa06e4df06711d00754a2f8a51d8b/Processor/Src/Pipeline/CommitStage.sv#L55-L77):
```verilog
module GetInsnPtr(
    output logic headOfThisInsn[2],
    input  logic last [2]
);
    always_comb begin
        for (int i = 0; i < 2; i++) begin
            headOfThisInsn[i] = 0;
            for (int j = i - 1; 0 <= j; j--) begin
                if (last[j]) begin
                    headOfThisInsn[i] = j + 1;
                    break;
                end
            end
        end
    end
endmodule
```

Code converted by sv2v: 
```verilog
module GetInsnPtr (
	headOfThisInsn,
	last
);
	output reg [0:1] headOfThisInsn;
	input wire [0:1] last;
	always @(*) begin : sv2v_autoblock_1
		reg [0:1] _sv2v_jump;
		_sv2v_jump = 2'b00;
		begin : sv2v_autoblock_2
			reg signed [31:0] i;
			for (i = 0; i < 2; i = i + 1)
				if (_sv2v_jump < 2'b10) begin
					_sv2v_jump = 2'b00;
					headOfThisInsn[i] = 0;
					begin : sv2v_autoblock_3
						reg signed [31:0] j;
						begin : sv2v_autoblock_4
							reg [0:1] _sv2v_jump_1;
							_sv2v_jump_1 = _sv2v_jump;
							for (j = i - 1; (0 <= j) && (_sv2v_jump < 2'b10); j = (_sv2v_jump < 2'b10 ? j - 1 : j))
								begin
									_sv2v_jump = 2'b00;
									if (last[j]) begin
										headOfThisInsn[i] = j + 1;
										_sv2v_jump = 2'b10;
									end
								end
							if (_sv2v_jump != 2'b11)
								_sv2v_jump = _sv2v_jump_1;
						end
					end
				end
			if (_sv2v_jump != 2'b11)
				_sv2v_jump = 2'b00;
		end
	end
endmodule
```

zachjs, 2022-05-01:
sv2v has logic to convert simple for loops in a Yosys-compatible manner. Although the inner loop here is sufficiently simple, sv2v was not identifying it as such. I have just pushed an update that should enable this input to be converted and ingested by Yosys. Please let me know if it works for you!

mmxsrup, 2022-05-01:
The code converted with the updated sv2v could be read by Yosys. Thanks for your quick support!

Newly generated code:
```verilog
module GetInsnPtr (
	headOfThisInsn,
	last
);
	output reg [0:1] headOfThisInsn;
	input wire [0:1] last;
	always @(*) begin : sv2v_autoblock_1
		reg [0:1] _sv2v_jump;
		_sv2v_jump = 2'b00;
		begin : sv2v_autoblock_2
			reg signed [31:0] i;
			begin : sv2v_autoblock_3
				reg signed [31:0] _sv2v_value_on_break;
				for (i = 0; i < 2; i = i + 1)
					if (_sv2v_jump < 2'b10) begin
						_sv2v_jump = 2'b00;
						headOfThisInsn[i] = 0;
						begin : sv2v_autoblock_4
							reg signed [31:0] j;
							begin : sv2v_autoblock_5
								reg signed [31:0] _sv2v_value_on_break;
								reg [0:1] _sv2v_jump_1;
								_sv2v_jump_1 = _sv2v_jump;
								for (j = i - 1; 0 <= j; j = j - 1)
									if (_sv2v_jump < 2'b10) begin
										_sv2v_jump = 2'b00;
										if (last[j]) begin
											headOfThisInsn[i] = j + 1;
											_sv2v_jump = 2'b10;
										end
										_sv2v_value_on_break = j;
									end
								if (!(_sv2v_jump < 2'b10))
									j = _sv2v_value_on_break;
								if (_sv2v_jump != 2'b11)
									_sv2v_jump = _sv2v_jump_1;
							end
						end
						_sv2v_value_on_break = i;
					end
				if (!(_sv2v_jump < 2'b10))
					i = _sv2v_value_on_break;
				if (_sv2v_jump != 2'b11)
					_sv2v_jump = 2'b00;
			end
		end
	end
endmodule
```

closed 2022-05-01 by mmxsrup